imurvai/brickcontroller2 #6: Sbrick powered by buwizz

dezsi-istvan, 2019-03-25:
If I power an sbrick from a buwizz (to controll more than 4 motors), playing can't start because sbrick need more time to power up.

imurvai, 2019-03-25:
Hi,
Is it a BuWizz 1 or 2? I have only a BuWizz 1 and it provides power to the
connected devices right after turning it on. On the other hand as I heard
from others the BuWizz 2 gives power only after connecting to it but I
can't confirm this behavior.
BR,
István

dezsi-istvan, 2019-03-26:
Hi,

Is a Buwizz2. I can confirm this behavior. When the green led blinking 
(search for connection) not provide power on outputs.

vicocz, 2019-09-02:
It somehow works for me when having SBrick powered by Buwizz2. It usually takes some time to connect, but in most cases AFAIK it's successfull.

closed 2020-05-08 by dezsi-istvan