imurvai/brickcontroller2 #8: The Smart HUB can not be controlled

paopao31224, 2019-05-03:
I found my smart HUB can only be connected, and the motor can not be controlled. The HUB is connected to POWERED-UP before using it. The Boost HUB can be used normally.

imurvai, 2019-05-03:
Hi,
I haven't experienced such an issue. I need to check whether there was a
firmware update recently for the PU hub and update the app if it's the
case..
BR,
István

paopao31224, 2019-05-03:
Hi, I am sure the firmware of the PU hub has been updated recently. Because I connected the PU app before using my new hub, then he automatically upgraded. Thank you for updating the app.

imurvai, 2019-05-05:
Hi,
After the firmware update I also lost the ability to control my PU device
with BC2. I'm working on the fix...

paopao31224, 2019-05-13:
> 嗨，固件更新后，我也失去了使用BC2控制我的PU设备的能力。我正在修复...
> [...](#)

Hi, I saw that you updated the code a few days ago. But the apps on the App Store and Play Now are not updated. I won't use VS, so is the problem solved now?

> Hi, After the firmware update I also lost the ability to control my PU device with BC2. I'm working on the fix...
> […](#)

Hi, I saw that you updated the code a few days ago. But the apps on the App Store and Play Now are not updated. I won't use VS, so is the problem solved now?

paopao31224, 2019-05-13:
> Hi, After the firmware update I also lost the ability to control my PU device with BC2. I'm working on the fix...
> […](#)

Hi, I saw that you updated the code a few days ago. But the apps on the App Store and Play Now are not updated. I won't use VS, so is the problem solved now?

paopao31224, 2019-05-13:
Hi, I saw that you updated the code a few days ago. But the apps on the App Store and Play Now are not updated. I won't use VS, so is the problem solved now?

hub has been updated recently. Because I connected the PU app before using my new hub, then he automatically upgraded. Thank you for updating the app.

imurvai, 2019-05-13:
Hi,
Yes, I've updated the code but it's not the final solution. Basically now the PUP hub is working but there is a significant lag. I'm working on the final fix but it will take time, so maybe I'll release a new version with the quick fix.

paopao31224, 2019-05-13:
Ok, I didn't think that a firmware update would bring so many changes. Thank you for your work, and I will wait patiently.

imurvai, 2019-05-13:
Yeah, unfortunately Lego has changed the PUP communication protocol that's why it stopped working in BC2

paopao31224, 2019-05-13:
I didn't expect a firmware change to change so much. Thank you for your work, and I will wait patiently.

vicocz, 2019-08-02:
Hi,
I have newly a Smart hub, but seems it's not possible to control it too using BC2 1.7 running on Android. Unfortunately I did not have chance to check it in any previous version, but a motor seems to work when using Powered Up app from LEGO.

imurvai, 2019-08-02:
Hi,
The recent changes should not cause it, I just added support for the new Technic Hub (42099).
Do you have the latest firmware on your Smart Hub? I just checked and mine is working ok.
Can the app see the device and connect to it BTW?

imurvai, 2019-08-02:
The advertisement data should be ok since your phone is able to discover and connect to the device. Tomorrow I'm gonna check the firmware on my PUP hub if it has the latest again and see how it goes...

vicocz, 2019-08-06:
Hey advertisment data from BLE sniffer running on the same Android phone as the BC2:
0x02 0x01 0x06 0x11 0x07 0x23 0xD1 0xBC 0xEA 0x5F 0x78 0x23 0x16 0xDE 0xEF 0x12 0x12 0x23 0x16 0x00 0x00 0x09 0xFF 0x97 0x03 0x00 0x41 0x07 0x00 0x63 0x00 0x05 0x12 0x10 0x00 0x20 0x00 0x02 0x0A 0x00 0x0A 0x09 0x53 0x6D 0x61 0x72 0x74 0x20 0x48 0x75 0x62

vicocz, 2019-08-08:
I was able to repro the issue in my environment (Windows desktop prototype of BC2).
BC2 was able to connect to PU and run an attached motor when having code base before #9. Once I rebased to the recent commit, able to connect but not run the motor.

imurvai, 2019-08-14:
Hmm that is interesting. The only change relating to the motor handling is that now the app sends the commands to a virtual port instead of the "real" ones to be able to handle the motors using one single command.

imurvai, 2019-08-14:
Could you check what is the virtual port ID given back by your PU HUB when asking for it? As far as I remember it is assumed to be 100.

vicocz, 2019-09-02:
It's send via SendOutputValueVirtualAsync which for PoweredUpDevice uses virtual port ID 0x10. The old way, when each channel is controllet separately, works.
The strange thing is that the same command via SendOutputValueVirtualAsync works properly for BoostDevice (Controlling ports A+B for LEGO Boost set 171001).

imurvai, 2019-09-02:
Hmm, that's wierd, using the virtual port is working for me, however BC2 only assumes that the virtual port ID is fixed to 0x10. I'll try to implement it properly...

andykrej, 2019-09-03:
Imurvai, Thanks for the great work!!
I'm playing with the PUP hub and motors from the batmobile set. It works fine with motors set to NormalMotor. But when set to ServoMotor, the motors don't turn. Is the servo function only supported on the bigger motors? (btw i'm using bc2 version from play-store)

imurvai, 2019-09-03:
Yes the new M motors used in the batmobil don't support servo mode. Only the new L and XL motors do.

vicocz, 2019-09-10:
> Hmm, that's wierd, using the virtual port is working for me, however BC2 only assumes that the virtual port ID is fixed to 0x10. I'll try to implement it properly...
It may be just issue with my device only.
The strange thing is: there is no Hub Attached I/O Message Type [0x04] comming and strangely there is a notification Error Messages Message Type [0x05] comming having this content: 0x05 0x00 0x05 0x81 0x06. According to LEGO Wireless protocol, it says the latest command 0x81 reports 0x06 | Invalid use (e.g. parameter error(s).

imurvai, 2019-09-11:
Hi, in the next release I'll get rid of the virtual ports. According to my testings all PUP device is working correctly using only normal port commands.

vicocz, 2019-09-19:
> 
> Hi, in the next release I'll get rid of the virtual ports. According to my testings all PUP device is working correctly using only normal port commands.

BC2 1.8 works propery for my PUP and M motor.  Good job @imurvai

imurvai, 2019-09-19:
Cool, good to hear that :)