imurvai/brickcontroller2 #11: Logitech F710 support

msdobrescu, 2019-08-21:
Hello,

Is it possible to use Logitech F710?
I'm not a console user at all, so I don't know how this work in the regard of pairing.
F710 comes with some dongle, I don't know if that is a problem, might it have its own protocol?
What would be your assessment?

Thank you.

imurvai, 2019-08-21:
Hi,
As I can see it's not a bluetooth gamepad but has it's own receiver using some propriatary communication protocol. Therefore it can only be used on PCs.
Try getting a bluetooth gamepad which can easily be paired with Android devices so that BrickController2 can also use it.

msdobrescu, 2019-08-21:
Hi, thanks for answering so fast!
Where do you see it's not a Bluetooth device? Is there a specification somewhere?
Also, do you have a list of compatible devices that you have tested?

imurvai, 2019-08-21:
Hi, I checked the specification sheet (https://www.logitechg.com/en-roeu/products/gamepads/f710-wireless-gamepad.html#product-tech-specs) and it doesn't mention it's a bluetooth. And usually gamepads (and other HID devices like mouses) having their own receiver are not bluetooth devices.
I don't have a list of compatible devices as I only have two bluetooth gamepads, a PlayStation 4 DualShock and a cheap gamepad from a chinese webshop.
Most bluetooth gamepads however are Android compatible and it's usually stated on the box, so you can safely buy almost any bluetooth capable gamepad.

msdobrescu, 2019-08-21:
I see, the frequency is misleading.
My issue here is that I can be tricked by shopping online and buy something not working because the specs are incorrect or partial often.

vicocz, 2019-09-10:
Some of cheap gamepads have their own USB dongle without supporting bluetooth radio at all. E.g. C-tech Lycaon is missing bluetooth radio (in comparsion to Evolveo F1 which looks like the same). There is a solution to connect it via USB OTG and it works both in Android and BC2 app.
Moreover it can be kinda workaround when your phone has issues when connecting several bluetooth devices (e.g. gamepad and 2 SBricks).

msdobrescu, 2019-09-11:
For the smartphone perspective, indeed, it is a solution, but is there possible to use a similar solution on BuWizz and alike?

imurvai, 2019-09-11:
What do you mean? Connecting BuWizz using OTG?

msdobrescu, 2019-09-11:
Yes. What's the point of connecting it to BC2 without making it work further with some LEGO controller?

vicocz, 2019-09-11:
> What do you mean? Connecting BuWizz using OTG?

It's misunderstaning. BuWizz/SBrick/PUP is still connected via bluetooth of course. I was talking about connecting a gamepad (which does not support bluetooth, but has some radio + own USB dongle) via USB OTG.
My phone from Xiaomi does not allow to have 3 concurrent bluetooth connections (e.g. gamepad + 2 BUWIZs), but 2 BT devices work fine.
Therefore gamepad connected via its radio USB dongle + USB OTG helps a lot. Or buing a new phone of course, but there is no benchmark which tells how many BT connection it allows. AFAIK hardware limit is 7.

msdobrescu, 2020-12-22:
Actually, F710 works fine if you have the right device, meaning one with USB support for it. With my Samsung T510 simply works, probably it has the driver, but my Nokia 7.1 does not. I think this ticket should be closed as dependent on the Android distribution.