imurvai/brickcontroller2 #12: Don't work with ipega PG-9156

MichelCibrario, 2019-09-02:
The Controller tester don't detect the controller ipega PG-9156 with:
SAMSUNG NOTE 9
SAMSUNG NOTE 3

imurvai, 2019-09-02:
Hi, did you pair the controller in the Android's bluetooth settings?

MichelCibrario, 2019-09-02:
Yes. It's paired to the phone with the bluetooth menu. The ipega works with the phone. Ps4 controller work fine.

imurvai, 2019-09-02:
Most bluetooth gamepads have a couple of operating modes which can be selected by pressing a button on it. It is depending on the model, so please check the ipega's manual how to change the mode. I have a cheap bt gamepad and it has 4 modes and only one is working as a gamepad.

MichelCibrario, 2019-09-02:
> Message du 02/09/19 20:10> De : "István Murvai" > A : "imurvai/brickcontroller2"
> Copie à : "MichelCibrario" , "Author" > Objet : Re: [imurvai/brickcontroller2] Don't work with ipega PG-9156 (#12)> >

IT work!!!!!! Mode home+X (standard mode) thanks !!!!
 
Cordialy