imurvai/brickcontroller2 #13: LEGO Boost Hub Via IOS and Nimbus SteelSeries controller

brickbuildmeister, 2019-09-09:
Thank you for providing this sweet app.  There is so much untapped potential using this tool.  I have been testing this set up for a few hours now.  I see (a), (b), (c), and (d) are all working fine via the test feature using the slider interface.  However, when i start assigning controls to various inputs on the wireless controller i can't control the (d) port.  I was wondering if its because the (a) and (b) are not controlled as a pair.  I have used the a,b,x,y, keys on the controller and i can move the (a) and (b) ports as a steerable vehicle with a working (c) port.   I just can't seem to add the (d) no matter if the other channels are assigned or not.   My goal is to unite the two (a) and (b) so that they can be controlled as one via a joy stick and be able to use port (c) and port (d). 
Environment info: LEGO Boost Hub Via iPad IOS and Nimbus SteelSeries controller.

imurvai, 2019-09-10:
Hi, to be hones I have never tried the D port on the Boost hub :) Today I'll check if it is working or not. Anyway I did some significant changes regarding the control of the new powered up devices, so the next release will contain some fixes.

brickbuildmeister, 2019-09-10:
Thanks for the response.  This is an amazing niche application.  Brickcon 2019 is happening 1st weekend of October and many folks use your app to control their creations.  Thanks again.

imurvai, 2019-09-10:
Wow that sounds great that so many people are using my app :)
And I just checked the Boost and I can control all its ports using the gamepad. At least in my dev version of the app. So in the next release all the ports should be ok.