imurvai/brickcontroller2 #15: Missing async API EnableNotification / WriteNoResponse

vicocz, 2019-09-21:
Hi,
first of all I really enjoy this amazing app therefore this is more API improvement suggestion.

Would it be possible to make these IBluetoothLEDevice methods EnableNotification / WriteNoResponse  async please:
`        Task<bool> EnableNotificationAsync(IGattCharacteristic characteristic);`
`        Task<bool> WriteNoResponseAsync(IGattCharacteristic characteristic, byte[] data);`

I work on porting this app to Windows by utilizing UWP but during adoption of the recent changes found out that my implementation of EnableNotification is locked when synchronously calls UWP bluetooth API [WriteClientCharacteristicConfigurationDescriptorWithResultAsync](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcharacteristic.writeclientcharacteristicconfigurationdescriptorwithresultasync).
When this call is made asynchronously it works. I'm neither aware of Android / IOS development neither have option to test them all, therefore fixing this by myself would not be safe for these platforms.
The second method WriteNoResponseAsync is just an improvement how to keep all write related methods async in general.

Best Regards
VicoCz

imurvai, 2019-09-21:
Hi,
I'm glad you like my app :)
I made the whole IBluetoothLeDevice API async for the sake of consistency. Anyway you can fork my repo any time and you can make any changes you want :)
BTW I've been also started creating a .net lib for the new PoweredUp devices. It won't be part of the BrickController2 repo, I'm planning to put it to GitHub separately and I'd like to cover all the features the PUP devices have in that lib like sensor notifications, virtual ports and so on...

vicocz, 2019-09-26:
Thanx a lot for the recent IBluetoothLeDevice async changes. Notifications are now working in my UWP prototype. Even I have my own fork, it's easier to keep in sync with yours.

> I've been also started creating a .net lib for the new PoweredUp devices. 

Sounds really interesting, I'll watch your repo as this was also my idea (sensors + voltage info) how to develop BC2 app in future. Especially when apps like BOOST deserve application to rival with :)

closed 2019-09-26 by vicocz