imurvai/brickcontroller2 #17: PoweredUp hub - Servo auto calibration internally fails due to IndexOutOfRangeException

vicocz, 2019-10-08:
There has been internal IndexOutOfRangeException in [ControlPlusDevice.OnCharacteristicChanged](https://github.com/imurvai/brickcontroller2/blob/master/BrickController2/BrickController2/DeviceManagement/ControlPlusDevice.cs#L159) due to reading of Int32 bytes outside the message data when trying to auto calibrate motor (defined as servo) connected on A port of PoweredUp hub. 
Seems it's just single byte missing.
Based on [Message format 0x46](https://lego.github.io/lego-ble-wireless-protocol-docs/#port-value-combinedmode) _Mode_ flag should be 16 bits and therefore _relPosition_ seems to be 16 bit only. Honestly, hard to say.
[image]

imurvai, 2019-10-08:
Hmm, strange I haven't seen such an exception. But I haven't tried the servo with the PUP hub, only with the Technic Hub. I'll look into it...
Thanks for letting me know.

vicocz, 2019-10-30:
I saw a recent improvement for this strange issue. I do not see the mentioned exception anymore!
Thx a lot!

imurvai, 2019-10-30:
That's good. Thanks for letting me know.

vicocz, 2019-11-03:
What is the next step for this item right now? Am I, as the reporter, expected to close it? Or use a label?

imurvai, 2019-11-03:
I think you can close it if you think it has been fixed.

closed 2019-11-03 by vicocz