imurvai/brickcontroller2 #20: Q: Buwizz2 channel swapping, is it correct?

vicocz, 2019-10-30:
I use this amazing BC2 to control my model driven by a Buwizz2 device. The strange thing is that my channels are swapped. E.g. if a connect XL motor to channel 3, it does not drive. But on Device page I'm able to drive it via 4th slider.
I know there is channel swapping code for Buwizz2 device. Mine device has following deviceData: 4e-05-42-57-00-1e. Therefore channel swapping is applied.

Attaching additional BLE data just in case:
[image]

imurvai, 2019-10-30:
Thanks for the BuWizz2 info. I added that channel swapping because several users reported swapped channels and the only thing that was different from other devices was the device data. But it seems it's sometimes misleading. Unfortunately there is no official documentation from the buwizz team so I can just guess in what cases should the channels be swapped.
BTW I don't have any buwizz2 so I can't really try.

vicocz, 2019-12-08:
You're right, unless there is either the specification or a reporter of the original channel swapping issue there is probably no proper solution.
Btw I wrote to their support, but as you said, there is no BUWIZZ protocol specification, just their open sourced app.
Additionally I was told they plan to cooperate with you and give better support to open source community in general. But it's just a promise.

imurvai, 2019-12-09:
Hi, I didn't know their app is open source. I'm going to search for it...
Yes, another guy also told me that the BuWizz team maybe wanted to contact with me about some cooperation but they haven't get in contact with me yet.
And I also asked them about the channel swapping but I haven't got any response.

vicocz, 2019-12-11:
Not sure if they meant exactly their Android app. I wrote to support@buwizz.com
Here is the part of the response:
> There is an open source BuWizz app release on GitHub:
> https://github.com/BuWizz/BuWizz-Android-Demo

imurvai, 2019-12-14:
I checked this project, but it doesn't contain anything relating to the channel order unfortunately. So this is still an open question...

vicocz, 2021-12-13:
At least data has been described in B2 API: `05:4E:’B’:’W’:’x’:’y’ in the application, where x and y are replaced with firmware version`
Channel swapping is decided by firmware version `0.30` (0x00-0x1e):
https://github.com/imurvai/brickcontroller2/blob/90918391f8a4c7c16b536f4a6171e49f4183b353/BrickController2/BrickController2/DeviceManagement/BuWizz2Device.cs#L45
It does not give any exact answer but IMHO once my B2 device gets higher version of firmware, it will not have swapped channels anymore.

imurvai, 2021-12-14:
Yes, unfortunately it is not mentioned anywhere from which version the channels are swapped. I tried to tweak the code based on user feedbacks, but it has never been 100% accurate.

vicocz, 2021-12-15:
Does it make sense to add per device setting for channel Buwizz2 swapping instead of detecion based on the firmware revision? please? At least my B2 has `0x1E` and therefore it swaps channels which is wrong in my case.

imurvai, 2021-12-15:
That's a good idea, I'll add it.

muellerj, 2023-06-24:
I just stumbled across the same issue and would very much appreciate an option to swap the ports on Buwizz 2.0 :)