imurvai/brickcontroller2 #32: Continued operation with the screen turned off

gath47, 2020-01-04:
Using a controller is great because it means I don't need to hold my phone, but since the app ceases to function when the phone screen is turned off it means that I have to pocket my phone with the screen on and can result in inadvertent inputs, etc.  How hard would it be to allow the app to run while the screen is off?

imurvai, 2020-01-04:
It was asked by other users too. Unfortunately the controller events are only dispatched to the application in the foreground. So whenever the app goes to background (including turning the screen off) the controller events doesn't get forwarded to the app.
What I was going to do is lowering the screen brightness during playing after a while. This unfortunately requires special permissions which is not supported by the library I'm using for handling the permissions. It's not forgotten though, I just put this feature on hold.

MoritzSGit, 2020-05-01:
Hi, I just wanted to mention you could use the proximity sensor to turn off the screen while the phone is in your pocket. There are also [3rd party apps](https://play.google.com/store/apps/details?id=ss.proximityservice) for this.

imurvai, 2020-05-06:
Good idea, I'll look into that option. Thanks.

kj-jozwiak, 2021-01-14:
Hi, sport travkers are using BT LE for sensor and are running in background, i'm using IpBike and it has a few modes of power saving. Another work around i saw is keeping persistent notification, not sure it will help, but maybe worth a try.

imurvai, 2021-01-14:
Hi, yes the BT connection is OK while in background but the controller events are only dispatched to the app when it's in foreground unfortunately.

kj-jozwiak, 2021-02-04:
Hi,
I'll try to find some time research and fix this is possible.
Is master branch the best to start?
Do i need anything else to run on my android device?
B.R.
Chris

imurvai, 2021-02-04:
Hi,
Yes, just clone the repo and create your own branch from master.
You need to set the developer mode on your phone and enable the 'usb debuggning' and 'install via usb' then select your phone as target in visual studio.
I'm curious what you will find...
BR,
István