imurvai/brickcontroller2 #33: Buttons vs Axis behaviour

paul-777, 2020-01-09:
On my controller some buttons e.g. Gas/Brake are recognised by the app as axis rather than buttons - which means if I try to use them as accelerators for a train the control isn't right. 
Let me explain, setup
Gas, PFI2,Red,Accelerator
Brake PFI2,Reg,Invert,Accelerator

Press Gas -> Fwd speed 1
Press Gas -> Fwd speed 2
Press Gas -> Fwd speed 3
Press Brake -> Reverse speed 3 (should be Fwd speed 2)
Press Brake -> Reverse speed 4
Press Gas -> Fwd speed 4

There is no way to slow down except using another button to issue stop
A similar issue exists when trying to use the hat buttons.
However if I use Buttons A and X instead of the Gas and Brake with the same settings it works as expected.

Seems like I need a way to customise the way the way the controller is interpreted to turn some "axis" into buttons.

thanks

imurvai, 2020-01-09:
Hi, yes unfortunately it is up to the Android system and the controller type to identify an input as an axis or a button, however it is possible to use the hat buttons for acceleration/deceleration similar to the joysticks.
The triggers are tricky, on some controllers they are buttons, but on others they are axises.

paul-777, 2020-01-09:
Thanks, would it be possible in the app to have settings to interpret certain axes as buttons - i.e. any non-zero number was interpreted as a button push rather than an axis movement? Thus allowing the broader range of Button Types to be used.

imurvai, 2020-01-13:
It's partially done for certain axis types, I'll see how I can extend it...