imurvai/brickcontroller2 #35: Duplo Train Base support?

michieldetailleur, 2020-04-25:
https://www.lego.com/en-be/product/steam-train-10874
https://www.lego.com/en-be/product/cargo-train-10875

Any plans for that? I read that they are also in the Powered Up family? And there are many projects around already that can connect to them:
https://github.com/nathankellenicki/node-poweredup
https://pypi.org/project/bricknil/

Thanks for the cool app!

imurvai, 2020-04-25:
Hi,
Actually I started adding the code for the Duplo Train but as I don't have such a motor I could not finish it properly. One of my users tested it and some basic functions are working but he said it is not perfect overall.
Anyway thanks for sharing that node based code, I'll try to borrow some ideas how to support the Duplo Train correctly :)

michieldetailleur, 2020-04-25:
Thanks! I can write a fair bit of Python, but unfortunately I don't have any experience writing Android apps. But I might just get started with it to help fix this :)

imurvai, 2020-04-25:
It's very easy to compile this app, you need to install Visual Studio (community edition is ok) with Xamarin support. Then clone this project, install the Android debug driver and enable developer mode on your phone. After that Visual Studio should see your phone once you connect to your computer. Run the Android project and it will compile, upload the app and start it.
If you uncomment line 86 in the BluetoothDeviceManager.cs file then the Duplo Train support will be activated.
It would be super cool if you could fix the duplo train issues.

michieldetailleur, 2020-04-25:
As always in coding, setting up the dev environment will probably be the hardest task :D I'll give it a try soon, but I first gotta play with my new Technic set (https://www.lego.com/en-be/product/4x4-x-treme-off-roader-42099) :) Your app is one of the reasons that won me over to buy this set! (and the fact I could get it at 10% discount :p )

imurvai, 2020-04-25:
Cool :) Let me know if you need help.

michieldetailleur, 2020-04-25:
Off-topic:
I just played around with the off-roader set, my PS4 controller and your app and I've gotta say: WOW!

Everything works just so intuitively! I got a simple "tank" steering trike up and running in no-time! The app just works sooo nice and smooth. I haven't even tried (or needed too) all the options and features, but I could totally see the usage for them immediately when glancing over them. It is so well thought out! Amazing and so much fun! Lego should just hire you to develop their PU apps!

imurvai, 2020-04-25:
Thanks, I'm glad you like it :)

michieldetailleur, 2020-04-25:
So, I was not wrong: being a total VisualStudio noob, setting up the dev environment IS pretty hard. It took me a few hours chasing dev kit and tools errors before finally managing to code, build and deploy a simple hello-world app to my phone.

Now I'm stuck with this when (re)building BrickController2:

Severity	Code	Description	Project	File	Line	Suppression State
Error	APT2260	resource style/Theme.AppCompat.Light.Dialog (aka com.scn.BrickController2:style/Theme.AppCompat.Light.Dialog) not found.	C:\Users\detailmi\Source\Repos\imurvai\brickcontroller2\BrickController2\BrickController2.Android\BrickController2.Android.csproj	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Xamarin\Android\Xamarin.Android.Aapt2.targets	160	

I did manage to "Restore NuGet Packages" and it says it's done them all, but no avail...

Any ideas?

imurvai, 2020-04-26:
Huh, I think it's gonna be some Android SDK issue. I'll look into it...

michieldetailleur, 2020-04-26:
I managed to fix it. Don't know exactly how :D 

I think I (re)installed Xamarin Forms using Tools -> Get Tools and Features.
And I manually installed Plugin.CurrentActivity using Manage NuGet Packages for Solution. The trick there however was to only install it for the Android subproject!

Now I'm struggling with deployment to phone. But I know it should work, it worked for my "hello world" app.

michieldetailleur, 2020-04-26:
Deployment worked after I disabled Use Shared Runtime (I was getting System.DllNotFoundException and that seemed to fix it).

But now when the app starts I get "Bluetooth devices will NOT be available". However my PS4 controller works. But I can't find my PU hub anymore (or the Duplo train :p). Bluetooth and GPS are enabled ofcourse.

I can find the string "Bluetooth devices will NOT be available" in the project. But I can't figure out what could trigger it.

imurvai, 2020-04-26:
The app requests permission for location access when it's first starting which is required for bluetooth operations. Did you enable that?

michieldetailleur, 2020-04-26:
Yes, I edited my answer, just when you replied :)

michieldetailleur, 2020-04-26:
It found the Train, hurray :) Now I can start testing :)

imurvai, 2020-04-26:
Cool, sounds good! :)

michieldetailleur, 2020-04-26:
Good news, it works perfectly!

I might try to add support for the sounds and the light. But just going forwards and backwards (with differing speeds, using the analog sticks) works perfectly!

michieldetailleur, 2020-04-26:
The direction seems to be wrong (so I have to select the inverted option). So I'll try and figure out that easy fix as a starter :)

imurvai, 2020-04-26:
try adding this method to the DuploTranHubDevice:

        public override void SetOutput(int channel, float value)
        {
            base.SetOutput(channel, -value);
        }

michieldetailleur, 2020-04-26:
That fixes it. However: now the right trigger on the PS4 controller (which in gaming land is by convention the forward throttle action) makes the train go backward and the left trigger makes it go forward.

Maybe all motors are "wrong" but this is only evident on the Duplo train where there is an inherent forward and backward direction of the platform? (i.e.: the light and hook indicate front and back)
As far as I can see the Lego Powered Up Train (https://www.lego.com/en-be/product/train-motor-88011) is also "directionless".

Could be tricky to fix this. Unless you break everyone's existing profiles :D Best to leave it like this then? Cause it will always be "wrong" for someone (stick vs trigger).

michieldetailleur, 2020-04-26:
> One of my users tested it and some basic functions are working but he said it is not perfect overall.

Can you expand on what was not perfect? Then I can test that a bit further. Cause for me it's fine. The basic forward/backward movement works. Everything else is nice to have and can be added gradually imho.

imurvai, 2020-04-26:
Ah I see what you mean. There is an invert switch on the controller action setting page with which you can change the direction the motor goes for the selected input. So on one of the triggers you need to invert the action.

michieldetailleur, 2020-04-26:
Indeed. But the code change just move the problem for the user:
* without the change: user has to use the invert switch on the left analog stick action
* with change: user has to use the invert switch on the right trigger action

Left analog stick forward and right trigger will always be opposite when using default setting (at least for my PS4 controller).
Also: left and right triggers will always be the same when using default setting. So for triggers you will always have to set the invert switch on one of them if you want to use them for forwards and backwards.

michieldetailleur, 2020-04-26:
Ok, so upon further testing I have found some differences with how the official Duplo Train app works.
When BC2 takes control of the train, the following stop working:
* push train in a direction to make it move in that direction
* "read" and act upon the colored rail inlays (eg: red makes train stop, white puts light on/off)

The overload stop still works however (if you send it into a wall with BC2 the overload protection still kicks in).

However, all this could be deemed an advantage too? :) For example when you have two trains, you could run one normally, controlled by the colored inlays, and one with BC2.

Or if BC2 could read the movement (front wheels) and color sensor you could reimplement these rules yourself in BC2 or implement new ones of your own!

imurvai, 2020-04-26:
Yes, by default the triggers move the motors the same direction. I think this is OK as is. It's working the same way for other motors too.
Also using BC2 and the official app at the same time could result unpredicted behavior as both apps can store their own states. And as BC2 doesn't use the sensors the two apps can fool each other.
So I advise not to use the two apps at the same time.
At the moment I'm not planning to add support for reading the sensors as it could make the app super complicated even for the users.
I started working on a new implementation by the way which aims full support for the control+ devices including automatic motor and sensor detection, full sensor support and so on...

zeraien, 2024-04-23:
Hi, is this abandoned? I thought maybe I'd try to take a crack at it?