imurvai/brickcontroller2 #37: Problem with 42099 and iPega 9118

vsvetoslavov, 2020-05-05:
Hi, I have iPega 9118 controller and connect it to my Xiaomi Mi 9T Pro phone. When controlling my 42099 I noticed at some point the steering gets stuck in one position (i.e. max right). When attempting to turn left, in some cases the steering goes to central position, and sometimes just nothing happens.

imurvai, 2020-05-06:
Hi, I've been reported such an issue from others but unfortunately I haven't seen it on my devices. What I can think of is for some reason the phone doesn't get the controller events sometimes, that is why it didn't go back. Please try to install the Bluetooth Auto Connect app, many times it fixes bluetooth related issues.

vsvetoslavov, 2020-05-07:
Hi
I believe it is not exactly problem with bluetooth, but instead... as if it recalibrates itself into rightmost (or leftmost) position. From then, when I press the joystick in the direction that is already at its end I often hear the beeping of the engine (overload alarm or something, not sure exactly). I will of course try this app.