imurvai/brickcontroller2 #38: Can't build project.

Kermit, 2020-06-15:
Hi.

I'm using Visual Studio 2017 Community and I can't build Android client.
I tried almost everything including upgrading all packages by NuGet. Maybe I have wrong .NET version?

`Can not resolve reference: `System.Threading.Tasks.Extensions`, referenced by `Microsoft.Bcl.AsyncInterfaces`. Please add a NuGet package or assembly reference for `System.Threading.Tasks.Extensions`, or remove the reference to `Microsoft.Bcl.AsyncInterfaces`.	BrickController2.Android`
`Can not resolve reference: `System.Threading.Tasks.Extensions`, referenced by `Autofac`. Please add a NuGet package or assembly reference for `System.Threading.Tasks.Extensions`, or remove the reference to `Autofac`.	BrickController2.Android`

Kermit, 2020-06-16:
I can compile project on commit f933b3dc9c646b427dd3efabb9a474ff60e006cf.
I assume I have some problems with Autofac greater than 4.9.4 but I don't know why.

imurvai, 2020-06-16:
Hi, hmm, that is strange, I haven't seen such an issue. Could you try with VS 2019?

Kermit, 2020-06-16:
I tried the VS 2019 yesterday and the compilation was successful but I couldn't run the app on my phone (Galaxy S8).
I read that you are building it on VS 2017 and I decided to try it too.

imurvai, 2020-06-16:
Ah sorry, I moved to VS2019 in the meantime. I forgot to update the readme file.
What is the reason you can't run it on your phone? Is it uploading to the phone?