imurvai/brickcontroller2 #42: How to connect to LEGO PoweredUp hub from 42099?

msdobrescu, 2020-12-22:
Hello, I try to connect to thePoweredUp hub from the 42099 set, but it seems not detecting it.
What is the procedure to connect to it? I try to understand if the hub is actually rejecting the connection.
BuWizz app can't detect it either.

msdobrescu, 2020-12-22:
OK, it works, it was some interference, I guess. BuWizz can't see it at all.
A question, now, as I see BC2 is a bridge between a hub and a controller physical device.
Are there plans to use BC2 itself to control with widgets, similar to PoweredUp, Controller+ or BuWizz apps?

msdobrescu, 2020-12-22:
I have tested further and I have:
- Samsung T510 which works fine with F710 but can't detect the hub in BC2.
- Nokia 7.1 which can detect the hub but can not work with T510, probably the driver is missing or something.
What can be done to make T510 detect the hub in BC2?
T510 works with the hub fine with Lego's dedicated apps.

msdobrescu, 2020-12-23:
Hello, I've reset the application data and works on T510.
The phone still can't access the controller.

Now, I try to configure the controller joystick to drive the car.
For PoweredUp motors, is there a way to reset the direction, like the Power Functions servo?

msdobrescu, 2020-12-23:
I have found the right combination, it seems that servo motor setting works for joystick to set the wheels back to straight direction.
What I can't do is to have a correlation between the joystick position and the wheels turn, i.e. for 50% of joystick range to have 50% wheels turn to the left or to the right. How can I achieve that?

imurvai, 2020-12-23:
Hi,
I'm glad you figured out in the meantime how to connect 👍 
What you can do in joystick mode is to set how much the motor should turn if you move the joy all the way left or right. By default it is 90 degrees but it can be set whatever you need between 0 and 180 degs.
For the on-screen controls (gadgets): The main reason I developed BC2 is to control Lego models using a game controller, and it would require a lot of time to add such a feature so I'm not planning to add it. Sorry.

msdobrescu, 2020-12-23:
Thanks, no need to be sorry, everybody has its goals.
You did a very good job that helped many to use something else than LEGO's software and hardware.
Regarding the joystick, I'd need more control. If it goes to 50% of the range, I guess it would be possible to position the motor as such, i.e. half of the range, from its origin.
Yesterday I have had a problem with Control+ that was unable to calibrate the direction anymore, hence my approach with BC2. I think would be useful such feature for your users.