imurvai/brickcontroller2 #59: Transferring creations to new phone ?

GrndTom, 2021-03-18:
Hi, thanks for the app ! How can I transfer a creation to a new phone ? Is there a file somewhere in the phone storage I can copy ?

imurvai, 2021-03-18:
Hi,
The creations are stored somewhere in a local sqlite database file. To be honest I never checked where these files are located. But if you are willing to build your own version check this branch: https://github.com/imurvai/brickcontroller2/tree/ProfileStorage
I'm working on a load/save mechanism for the profiles in this branch. It's working now but I'm still testing it before releasing.

jdebardi, 2021-03-18:
For iOS storing the config in iCloud would be ideal. That way I can have my creations sync across my devices. I regularly use 2 different iPads and my phone and have to build everything multiple times.

imurvai, 2021-03-18:
Yeah, I know iCould would be the best and googledrive for android likewise, but it's a bit trickier to implement.