imurvai/brickcontroller2 #60: Buwizz 3.0 support

10on, 2021-07-08:
Hi, thanks for your app. It’s really cool. I’ve buwizz 3.0 and basicly programming skill (I works as web developer). I can help for add buwizz 3.0 support, but don’t know what need do for it. I can be helpful for this project?

smadschm, 2021-09-26:
Meanwhile the bluetooth protocoll for BuWizz 3 are published.
https://buwizz.com/BuWizz_3.0_API_3.6_web.pdf
Would be great, if imurvai could add BuWizz 3 to this app.

imurvai, 2021-09-27:
Hi,
Yes, the protocol has been finally published :)
I've already started working on adding support for BW3 in this branch: https://github.com/imurvai/brickcontroller2/tree/BuWizz3Integration
You mentioned you are willing to help me. As I don't have any BuWizz3 can I ask at some point to test the app?

10on, 2021-09-27:
Yes, I'll glad to help you.
I've one buwizz 3.0 device, android and ios phones and different lego electrics.

imurvai, 2021-09-27:
Hi, In theory I've added the basic usage of buwizz3 into the app. Can you compile and deploy it to your phone?
Alternatively you can download the apk from here: https://www.dropbox.com/s/kcmub79cgijvdkw/BrickController2_2_9.apk?dl=0

10on, 2021-09-27:
I can try it tomorrow!
Thanks!

10on, 2021-09-28:
Buwizz 3 found via bluetooth.
But I have error on connect.
[image]

imurvai, 2021-09-28:
Thanks for the feedback. It's a good first step that the app can find the BW3 at least :)
Most probably it doesn't find the bluetooth service I specified.

imurvai, 2021-09-28:
I built another one with some fixes: https://www.dropbox.com/s/hgw6si2qj45hwlr/BrickController2_2_9a.apk?dl=0

10on, 2021-09-28:
Connected! 
But no any reaction on interaction with sliders.
I tried connect different electrics
[image]
[image]

imurvai, 2021-09-28:
Thanks for trying it out. Maybe it requires some additional setting up after connection. I'll try to figure it out from the documentation.

imurvai, 2021-09-30:
Hi, I found something that might caused the motors not working and fixed that in this build: https://www.dropbox.com/s/ueu9w9exz8iqcjr/BrickController2_2_9c.apk?dl=0
Could you try it out?

10on, 2021-09-30:
Reaction on sliders - perfect!
All six ports woks - PF and PU.
Tested with leds and motors. 

On nearest weekend I should be can test app with gamepad and all settings (stepper, servo etc).

if you give recomendations - will be good.I'm not good test engeneer :)

PS - Not displayed battery level and HW/FW version.

imurvai, 2021-09-30:
Hi, that's a super good news, thanks for testing. However it is only the basic functions I've implemented so far, so only the normal motor functions are working for PU motors (no servo or stepper yet). Of curse you can use the PF servo instead.

smadschm, 2021-10-06:
Is there some progress? I am sorry to not volunteer as a tester (i dont want to install an apk). So i'd like to know when the app in the store could be updated for the buwizz 3.

imurvai, 2021-10-06:
I think I can upload the app to the store in the weekend or early next week. But be aware that the BW3 support is not complete, the PU motors work only in normal mode yet.

DenoG1997, 2021-11-26:
Hey everyone 
is tehe any progress for teh bw3 PU servo support? :)

vicocz, 2021-12-06:
Tried to add support based on API doc and experiments, see #74 .
When action is configured as SERVO, PU absolute positioning mode is used.  For STEPPER relative position is initialized after connection and increased. Both requires proper PU motor (e.g. not M one).
Still not sure if API is used properly as servo/stepper configured motors are whistling little bit. And possibly stepper initialization seems to be still fragile.

DenoG1997, 2022-01-28:
How can I get access to TestFlight to test the servo for the Buwizz 3.0?

imurvai, 2022-03-23:
Version 3.1 is on TestFlight now.

DenoG1997, 2022-03-23:
it works fine on my Buwizz 3👌🏽

imurvai, 2022-03-23:
That's great to hear. Thanks for the feedback.

smadschm, 2022-03-23:
Thank you very much. Your effort is appreciated.

BobbyGraph, 2022-04-13:
> Version 3.1 is on TestFlight now.

Hi! Possible to have acces to apk for Android for testing? Thanks 👍🏼

vicocz, 2022-04-25:
This should be part of the recently released app version 3.1, shouldn't it? The only thing I see to be done is missing update of `README`.

BobbyGraph, 2022-04-25:
> This should be part of the recently released app version 3.1, shouldn't it? The only thing I see to be done is missing update of `README`.

Yes 3.1 is on Google Play thanks

PR #83 (2022-04-26): Add BuWizz 3 as supported device
This is IMHO the last piece to close #60.
commit 15979b65 (2022-04-26): Add BuWizz 3 as supported (#83)

This is IMHO the last piece to close #60
closed 2022-04-26 by imurvai