imurvai/brickcontroller2 #76: Connection Issue, cant connect buwizz with xbox controller connected, app goes into connection loop

mcneds, 2021-12-27:


imurvai, 2021-12-27:
Hi, does the same happen when you disconnect your xbox controller? Is it a BuWizz 1, 2 or 3?

mcneds, 2021-12-27:
No it only happens with it connected and all of the buwizzes are 2. It goes
into a boot loop of one connecting then immediately disconnecting and then
the other one does the same.  It also does the same thing with different
controllers

vicocz, 2021-12-28:
In past I had similar issues when I was not able to connect 2 Buwizz2 and a blutooth gamepad. There was nothing wrong with the app but my Xiaomi phone was able to keep 3 bluetooth connections at time, Sometimes it helped to reset bluetooth or restart the phone. Therefore I believe your situation could be the same. I recommand to check another phone.

imurvai, 2021-12-28:
Yes, I was thinking of the same. Another option could be to check if there are another bluetooth devices around (and maybe connected to your phone) and turn them off to reduce the number of active devices. It could be problematic for certain phone models.

mcneds, 2021-12-28:
Ok, I'll check it out.
I’m at my grandparents right now so try later today when I get home. Thanks
for the tip!