imurvai/brickcontroller2 #78: My Buwizz v1 device is not able to keep constant channel output for longer period

vicocz, 2022-01-07:
I've got Buwizz v1 device recently and it has following strange behavior: If there is constant motor output (e.g. via simple toggle or just during testing of channel output), the output is stopped after circa 3 seconds unless a channel change triggers another output write. The original Buwizz v1 app works fine.
Hard to decide whether it's bug in BC2 or just a kind of malfunction of my device.
Could someone, who has V1 device, confirm it please?

DenoG1997, 2022-01-28:
@vicocz @imurvai  how can i get access to TestFlight because I want to test the servo function for the Buwizz 3.0

imurvai, 2022-01-28:
ah, sorry, I haven't had time yet to upload the recent changes to testflight

DenoG1997, 2022-01-28:
No problem but I don’t even know how to install it on my iPhone even if you upload it

DenoG1997, 2022-01-28:
Can you tell me what Ive to do?

imurvai, 2022-01-28:
As far as I know once I add you to my testfligh then it will let you know when a new app is ready to be tested. So could you send me your email address?

DenoG1997, 2022-01-28:
Yes great ! caan97@hotmail.de

imurvai, 2022-01-28:
Thanks, I added you to the testers.

DenoG1997, 2022-01-28:
[image]

this is my recent fastest and best car with buwizz 3.0

DenoG1997, 2022-01-28:
but it is so fast that it very hard go control ist with the touchpad when you full buwizz support it than i can control it mich vetter with an controller

DenoG1997, 2022-01-28:
im studying now and next semester i will learn C an C++ i will can help you to programm

PR #79 (2022-09-29): Add output write expiration for Buwizz V1 devices
Added output write expiration (1500 ms) so as in case there is no channel output change there is still regular output write so as it resolves #78.
Additionally fixed minor issue change of output level has not been propageted until there has been some channel output change.
closed 2022-09-29 by imurvai