imurvai/brickcontroller2 #80: Bluetooth permisson problem (Android)

mwinkler, 2022-04-23:
Hi
In the latest version (v3.1), the application is not working properly anymore on my phone (Pixel 3, Android 12).
Scanning or connecting to devices (Buwizz, Technic Hub) abort with a error.

I debugged the application and got the following exception (when calling `BluetoothLEService.NewScanAsync(...)`):
`Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@6b9584d5: GattService registerScanner`.

I was not able to fix the problem, so maybe you can have a look at it.
Thank you

imurvai, 2022-04-23:
Hi, that is strange. I'll debug it and push a fix as soon as I can.
Thanks for letting me know.

imurvai, 2022-04-25:
The fix has been  released.

mwinkler, 2022-04-25:
@imurvai Works perfect now. Thank you very much for the fast fix.

closed 2022-04-25 by mwinkler