imurvai/brickcontroller2 #81: Can't connect to Powered Up Technic Hubs

ValouFCH, 2022-04-23:
Hello,

I've used the app for a while with a Buwizz 2, and it worked perfectly fine.

Recently, I got the 42100 set, and wanted to try the Technic Hubs with BC2

However, when I try to find them in the app, they are not listed. 

They connect fine to the officials Lego "Control +" and "Powered Up" apps, and I can see it listed when I search via my phones bluetooth settings.

One major Difference I saw is that both officials Lego App ask for Location access, so I tough that maybe I forgot to accept it in BC2, but the app doesn't acually ask for Location access so I can' give that permission.

I have an OPPO A74 5G 
Running Android 11
Color OS 11.1

What could I try ?

hlavoj, 2022-04-24:
I have same problem. 
Phone: Pixel 4
Android 12

Lego models: 42099, 42109

imurvai, 2022-04-24:
Hi,
I'm facing the same issue on my new Android 12 device. Started working on a fix...

ValouFCH, 2022-04-24:
Last update fixed it, I managed to connect Technic Hubs flawlessly :D

closed 2022-04-24 by ValouFCH