imurvai/brickcontroller2 #85: Q: Migration from Xamarin 5 to .NET MAUI

vicocz, 2022-05-10:
Hi,
do you have any plans to migrate from Xamarin.Forms 5 to newly developed .NET MAUI? Currently still in preview, but IMHO this is the future UI for multiplatform apps.
E.g. migration guide: https://github.com/dotnet/maui/wiki/Migrating-from-Xamarin.Forms-(Preview)

imurvai, 2022-05-11:
Hi, yes  definitely. I lokked into it a bit but didn't eant to start thr migration until it has been released

vicocz, 2022-05-24:
It have been released, 👀 
https://github.com/dotnet/maui/releases/tag/6.0.312

imurvai, 2022-05-24:
Wow, cool. Thanks for letting me know :)

vicocz, 2023-11-20:
Hi, @imurvai, I've started porting to .NET 7 MAUI in my fork, seems I've reach basic factionality of Android app.
Is this something you would be interested to have it merged here? If so, I can create DRAFT to share the state of migration.

vicocz, 2023-12-01:
DRAFT (to track my progres) is here: https://github.com/imurvai/brickcontroller2/pull/101

vicocz, 2024-04-10:
@imurvai Support for Xamarin Forms 5.0 is ending soon (on May 1, 2024)
https://dotnet.microsoft.com/en-us/platform/support/policy/xamarin

Do you plan to migrate the project to .NET MAUI?

imurvai, 2024-04-10:
hey, yes, but unfortunately I can't promise when I'll have time for that.

vicocz, 2024-06-19:
Resolved by https://github.com/imurvai/brickcontroller2/pull/115

closed 2024-06-19 by vicocz