imurvai/brickcontroller2 #91: Support for Mould King Hub

ernprivado, 2022-12-18:
Hi I notice that there is no support for mould king hub. I was just wondering if there is a plan for it

imurvai, 2022-12-18:
Hi, unfortunately I don't have any mould king device, so I can't add support for it. But the repo is open, so if someone implements it I'll release a new version :)

J0EK3R, 2023-09-18:
Hello :)

I just sniffed some BT LE telegrams sent by the Mould King App with the free Android App "nRF Connect" .

It seems that the App from Mould King (and the others from China) simply send Advertisement telegrams to their Hub-Bricks...

You don't need to have any Hardware to sniff the telegrams - just install the App und start sniffing...

Do you think it would be possible to integrate a BT-Device in your App that sends predefined Advertisement telegrams?
How and where would you start implementing such acdevice in your Code?
I'm able to compile your projects to a running APK with VS Community Edition.

J0EK3R, 2023-09-30:
I've forked your repo and added a new branch "BluetoothLEAdvertisingDevice".

https://github.com/J0EK3R/brickcontroller2/tree/BluetoothLEAdvertisingDevice

There is a new base class "BluetoothAdvertisingDevice" for "chinese" devices wich are controlled over one-way Bluetooth Advertising Messages.

I've implemented two devices:
* [MouldKing Robot](https://www.amazon.de/dp/B09TPHJ5L7?psc=1&ref=ppx_yo2ov_dt_b_product_details)
* [HOGOKIDS Robot](https://www.amazon.de/dp/B0C9Q61RJ2?psc=1&ref=ppx_yo2ov_dt_b_product_details)

Till now - for proof-of-concept - if you start a scan in the app bouth devices were inserted.
Then you can handle them like the others and bind channels.

[image]

regystro, 2024-01-10:
That's awesome!!

I'd like to try it with my Mould King Robot. Would it be possible to merge this into a new version of the app?

arthurson, 2024-03-27:
> I've forked your repo and added a new branch "BluetoothLEAdvertisingDevice".
> 
> https://github.com/J0EK3R/brickcontroller2/tree/BluetoothLEAdvertisingDevice
> 
> There is a new base class "BluetoothAdvertisingDevice" for "chinese" devices wich are controlled over one-way Bluetooth Advertising Messages.
> 
> I've implemented two devices:
> 
> * [MouldKing Robot](https://www.amazon.de/dp/B09TPHJ5L7?psc=1&ref=ppx_yo2ov_dt_b_product_details)
> * [HOGOKIDS Robot](https://www.amazon.de/dp/B0C9Q61RJ2?psc=1&ref=ppx_yo2ov_dt_b_product_details)
> 
> Till now - for proof-of-concept - if you start a scan in the app bouth devices were inserted. Then you can handle them like the others and bind channels.
> 
> ![Screenshot_2023-09-30-06-28-41-055_com scn BrickController2](https://private-user-images.githubusercontent.com/24459441/271753669-324ecc47-e817-4002-b685-aa50e9e09326.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE1MzQ0MDEsIm5iZiI6MTcxMTUzNDEwMSwicGF0aCI6Ii8yNDQ1OTQ0MS8yNzE3NTM2NjktMzI0ZWNjNDctZTgxNy00MDAyLWI2ODUtYWE1MGU5ZTA5MzI2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI3VDEwMDgyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2ZDA1NzdjYTM5YjEyYjI2YWIzZjJjYzc3MjRjNTJhYjQ5NTQ0OTczNjMxNzcyNTg0ZDcwYzViZTYyYzI5NzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.504ZjUKhqt5u48WvsJGMWnlgH4HNHAANYLkEkihQ7R8)

can you give us app so we can try

J0EK3R, 2024-04-03:
Here ist the try:
Sorry, Android only: [Release J0EK3R3.3](https://github.com/J0EK3R/brickcontroller2/releases/tag/J0EK3R3.3)

arthurson, 2024-04-03:
> Here ist the try: Sorry, Android only: [Release J0EK3R3.3](https://github.com/J0EK3R/brickcontroller2/releases/tag/J0EK3R3.3)

just tried, always show Bluetooth devices will not be available, i can connect my radio through Bluetooth, but i cant connect mk4.0, the apps shows tons of options btw,  i tried all without luck

[image]

arthurson, 2024-04-03:
i can only use old black moukking 2.2 to connect mk4.0 at the moment 
https://youtu.be/I4xvohZ80MI?si=piuzx8aMSXQ7fo4S

regystro, 2024-04-03:
> Here ist the try: Sorry, Android only: [Release J0EK3R3.3](https://github.com/J0EK3R/brickcontroller2/releases/tag/J0EK3R3.3)

I just tested MouldKing 6.0 Module and seems to work pretty well. Also connecting same hub as MoludKing Mecanum Module, it works as expected.

J0EK3R, 2024-04-03:
> > Here ist the try: Sorry, Android only: [Release J0EK3R3.3](https://github.com/J0EK3R/brickcontroller2/releases/tag/J0EK3R3.3)
> 
> just tried, always show Bluetooth devices will not be available, i can connect my radio through Bluetooth, but i cant connect mk4.0, the apps shows tons of options btw, i tried all without luck
> 
> ![20240403_135724](https://private-user-images.githubusercontent.com/110549393/319062147-659a3e97-169a-4a1f-b6ea-25118f10a082.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIxMzMyOTgsIm5iZiI6MTcxMjEzMjk5OCwicGF0aCI6Ii8xMTA1NDkzOTMvMzE5MDYyMTQ3LTY1OWEzZTk3LTE2OWEtNGExZi1iNmVhLTI1MTE4ZjEwYTA4Mi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwM1QwODI5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yOGI5NjhkNzJmY2UwMzA3MWE5NTIzNTJkOTFhZDA0Mzg4NmNhM2YwZWM4ZTA5MDJiMzExYTA5ZjlmZDA2MTQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.v5eM94KmmVHNbD_p4kFioRuiALB8RCqpJDoy0O0vQe8)

Did you give all Bluetooth rights to the app?

arthurson, 2024-04-03:
> Did you give all Bluetooth rights to the app?

i installed to 2 android,  different results
both dont ask bt rights but location and storage rights only, like original app
on android 10, it shows few options, no pf1~pf4, it says connected but no response on hardware when choosing mk stuff
on older tablet samsung tab s based on android 6, it has pf1~pf4, and it says failed to connect
i will try another phone for test

J0EK3R, 2024-04-03:
pf1-pf4 is the infrared part - that plays no role with bluetooth.
When starting the app for the First time I grant every rights and it works...

arthurson, 2024-04-03:
> pf1-pf4 is the infrared part - that plays no role with bluetooth. When starting the app for the First time I grant every rights and it works...

maybe  mk4.0 and mk6.0 has different setting. like when i chose mk6.0 to connect mk4.0 battery on old black mk app , it wont works as well.

it seems you are the only person who knows how to make mk stuff work on this app at the moment, will you implement mk4.0 as well?  like 13107 and 17036, these products use 2 batteries to handle 8ch at the same time, if we can use better radio to play with it, it will be a big surprise to mk users

J0EK3R, 2024-04-03:
...if I find some time - and that's the problem ;) - I will try to add support for MK4.0.
I just ordered the MK4.0 from the China man.
Stay tuned on this channel! :)

J0EK3R, 2024-04-03:
...if I find some time - and that's the problem ;) - I will try to add support for MK4.0.
I just ordered the MK4.0 from the China man.
Stay tuned on this channel! :)

But the next problem would be how to differ between the two Bluetooth-Receiver-Bricks, hmmm...

J0EK3R, 2024-04-08:
Here ist another try [3.3_J0EK3R_Beta_2](https://github.com/J0EK3R/brickcontroller2/releases/tag/3.3_J0EK3R_Beta_2).

I tryed (without having such devices) to implement (3 devices of) MK4.0 and (3 devices of) MK6.0.

**MK6.0:**
To bind _MK6.0-Device 1_ to DeviceA in the app just go to the device-page, select **MK 6.0 - Device A** and press the button on the MK6.0 brick. Test the channels.
Same for the second and third MK6.0.

**MK4.0:**
There is just one device for the MK4.0 in the app - but with 12 channels.
The first 4 channels are for device 1, the second 4 channels for device 2 and the third 4 channels for device 3.
I don't know how to bind the devices? - Are the MK4.0-bricks hard-coded to device 1, 2 and 3?

arthurson, 2024-04-08:
just quick tested, now the app can connect 2 mk4.0, 8ch works at this test page
[image]

yet i can't select the ch on this page so my rc controller can't test, normally i should choose the ch i want.
[image]

[image]

J0EK3R, 2024-04-08:
Puuuuh - weird ;)
I take a look....

J0EK3R, 2024-04-08:
Next try:
[3.3_J0EK3R_Beta_4](https://github.com/J0EK3R/brickcontroller2/releases/tag/3.3_J0EK3R_Beta_4)

arthurson, 2024-04-08:
[image]
now ui shows 1-12,  1 is red, but actually i cant control ch1 by my rc radio, and i cant choose other ch to test

J0EK3R, 2024-04-08:
Nice video :)
Tomorrow I will look for the UI problem...
On my mobile or my tablet I have more pixels.

All the channels of the two MK4.0 are working, right?

arthurson, 2024-04-08:
> Nice video :) Tomorrow I will look for the UI problem... On my mobile or my tablet I have more pixels.
> 
> All the channels of the two MK4.0 are working, right?

yes, the app will connect 2 batteries at the same time after presing scanning, both are ch1-4,  pressing hard button once on battery will switch to 5-8, just like mkapp

J0EK3R, 2024-04-09:
Next day, next try ;)
[3.3_J0EK3R_Beta_5](https://github.com/J0EK3R/brickcontroller2/releases/tag/3.3_J0EK3R_Beta_5)

J0EK3R, 2024-04-09:
Now with Button "Delete All Devices"
[3.3_J0EK3R_Beta_6](https://github.com/J0EK3R/brickcontroller2/releases/tag/3.3_J0EK3R_Beta_6

arthurson, 2024-04-09:
now i can select the ch, but my rc radio cant control it, no matter i use bt or usb cable

every beta make a good step progress 

https://youtu.be/1XeEG6RoC70?si=1HmMjP979_CX7u9s

J0EK3R, 2024-04-09:
After binding a controller action to a channel you have to press the OK-button on right top to save that.
Then - after binding all controller actions to channels - you have to  **RUN** your **Controller-Profil** or **Creation** by pressing the **Play-Button** on right top.

J0EK3R, 2024-04-09:
I think there ist a Bug with the negative direction in the MK4.0 Module:
If you pull the Slider to the left - negative direction - the connected motor starts with full Speed and decreases speed when the Slider moves more left.
Do you have the Same behavior?

arthurson, 2024-04-09:
> I think there ist a Bug with the negative direction in the MK4.0 Module: If you pull the Slider to the left - negative direction - the connected motor starts with full Speed and decreases speed when the Slider moves more left. Do you have the Same behavior?

yes, the curve is nonlinear. if you set ch from -100 to100, one direction will not be activated, like -70 to 70 will be ok for both sides

J0EK3R, 2024-04-09:
Now it's fixed:
[3.3_J0EK3R_Beta_7](https://github.com/J0EK3R/brickcontroller2/releases/tag/3.3_J0EK3R_Beta_7)

arthurson, 2024-04-10:
beta8 tested, it seems you reduce the output to 70% so gimbal can control both direction.

my toy 17036 backhoe use clutch to change gear, i tested it for  long run, the motor brunt after 10 mins, the design of that part is not relible. now i need to find a replacement.

 i found even the ble joystick has 16ch signal, only 8 ch are proportional, throttle is 0-1, that is no enough in my case. it  use 10ch to fullfil all the actin. need some time to change settings 

also i ddnt test the setting the app given, i use my radio to set, like dead point, end point and cruve etc.

arthurson, 2024-04-10:
https://youtu.be/kk-63aSYCuA?si=gamkOwaD_-rchZM8

now the backhoe can run as it should be, except i need to change the gear manually as motor broken yesterday, it really prove that we should have a good controller for the better handling.

thx now i am satisfied

arthurson, 2024-04-10:
it is purely a moc design default, mouldking dont use clutch wheel to protect the motor, it makes the motor overheat when it keeps on pushing one direction, motor died as result.

if you wanna know how they design the gear , find the 17036 manual, b1 motor is for changing gear.

it is well-known that mould king always steal moc without any credit, their excavator crane backhoe are no exception. this complicated product actually come from this designer.

https://www.moc-builder.com/jcb-backhoe/

i will teardown this L motor for repairing tomorrow.

J0EK3R, 2024-04-10:
And is the problem with the **MK4.0 implementation** fixed with the **Beta 8**?

arthurson, 2024-04-10:
ive posted this beta on fb group, hoping more people to test it, here is the link 
https://m.facebook.com/groups/mouldkingfans/permalink/1136277217565127/?mibextid=Nif5oz

arthurson, 2024-04-10:
to repair the motor ive already teardowned my car, no further test can be done at the moment, for me beta8 is quite successful, just wait other people to find bugs

J0EK3R, 2024-04-14:
I did some beautifying and little fixes.
Here is another try:

[3.3 J0EK3R Beta 9](https://github.com/J0EK3R/brickcontroller2/releases/tag/3.3_J0EK3R_Beta_9)](https://github.com/J0EK3R/brickcontroller2/releases)

elclaudio, 2024-05-11:
I would love official support for mould king hub (especially the 6.0 which has 6 ports !) I have a few of them,all is ok but now I see that for serious stuff, I need programability with sensors (simple switchs at first would be enough). 

For this, I was thinking about using a cheap raspberry pi pico W to manage the sensors via gpio (switch only) or better via I2C (TCS34725 RGB Color Sensor, TOF10120 distance sensor, MCP23008 or MCP23017 for input switch...) the pico then receive the input from the sensor and send the propers orders through bluetooth to drive the PF motors, servos, light etc via  the mould king hub. The python program would drive the motors accordingly.

All we need is already here (the mould king protocol), but we need a library in python to drive the mould king hub. Could this be done ? may be @J0EK3R could help ? this would be fantastic because Lego spike hub and sensors are insanely pricey and to big

@J0EK3R Is it possible to isolate the mould king part of the app to make such a library ?

J0EK3R, 2024-05-12:
Hi @elclaudio,
your project sounds very interesting and i think it would be possible...

The "mould king part" is already separated - look at
* [MouldKingCrypt.cs](https://github.com/J0EK3R/brickcontroller2/blob/master/BrickController2/BrickController2/DeviceManagement/MouldKingCrypt.cs) for the encryption part
* [MouldKingBaseByte.cs](https://github.com/J0EK3R/brickcontroller2/blob/master/BrickController2/BrickController2/DeviceManagement/MouldKingBaseByte.cs) is the baseclass for the MK 6.0 module
* [BluetoothAdvertisingDevice.cs](https://github.com/J0EK3R/brickcontroller2/blob/master/BrickController2/BrickController2/DeviceManagement/BluetoothAdvertisingDevice.cs) sets data to be bluetooth advertized...

Sadly the code isn't very cleaned up and optimized but with some comments and self speaking... ;)

I don't know if your platform supports that kind of advertising telegrams.
Android works very well, i.e. Windows doesn't - till now, I wasn't able to send the needed bluetooth advertising telegrams on Windows because of windows limitations.
I haven't tested other platforms yet like IOS or Linux.

elclaudio, 2024-05-12:
@J0EK3R thanks you for your prompt response. Yeah, the problem right now is that there is no "HAT" for raspi or other SBC to drive PF motors (only exists the build HAT to drive new lego powered up motors and even then there is only 4 ports, non stackable which is clearly not enough. The advantage here with mould king is that you can drive at least 3 hub with 6 PF ports per hub = 18 motors ! + possibility of stacking) 
While looking at the code, I've found theses files indeed. But I'm not familiar with coding, I've done a little bit of C# but not enough to start something like this, at least not without serious help. 
The plateform to be used is not written in stone, we can use whatever works best. I've talked about the pico because it cost nothing, have many gpio and i2c, uart etc and power consumption is lower than a regular pi with linux. Sadly, I've 2 pico right now but without bluetooth... I could buy one to test naturally. OR, I could try using a raspberry pi zero W which have bluetooth and running linux.
So basically, first, we need some very basic code (either python, C whatever) to connect to the mould king hub and eventually send motor ON / OFF commands to test feasibility. Compatibility with windows is not required since the plan is to run it on either a micro controller (pico) or very light hardware like pi zero W

J0EK3R, 2024-05-13:
Hello @elclaudio,
I'm very sorry but I have no experiance in python yet - but the main problem is: I have no time for such a project.
If somebody was porting the code to a python lib I could perhaps help if there were problems but... sorry, I have no time at the moment... :|

elclaudio, 2024-05-13:
> Hello @elclaudio, I'm very sorry but I have no experiance in python yet - but the main problem is: I have no time for such a project. If somebody was porting the code to a python lib I could perhaps help if there were problems but... sorry, I have no time at the moment... :|

I understand. But may be can you just help me with sending the command under linux shell. I tried with nrf connect, cloning some commands, it works but I haven't managed to make it work with linux commands : 

for exemple, the telegram to connect:
bluetoothctl
advertise menu
manufacturer 0xFF00 0x6D 0xB6 0x43 0xCF 0x7E 0x8F 0x47 0x11 0x88 0x16 0x7B 0x38 0xD1 0x7A 0xAA 0x26 0x33 0xC2

then what ?

or with hcitool : 
sudo hcitool -i hci0 cmd 0xFF00 0x6D 0xB6 0x43 0xCF 0x7E 0x8F 0x47 0x11 0x88 0x16 0x7B 0x38 0xD1 0x7A 0xAA 0x26 0x33 0xC2
I don't understand the usage and meaning of ogf, ocf
        cmd <ogf> <ocf> [parameters]

I miss something... but I believe I'm close. If we can send such commands directly to drive the motor (working with nrf connect) a simple shell script could do the job

J0EK3R, 2024-05-14:
Hi @elclaudio :)
I'm absolutely no BT-expert and I also did not manage to send the correct advertisement telegrams with bluetoothctl on my raspi.

Btw.: the manufacturer Id is 0xFFF0

`
bluetoothctl
advertise menu
manufacturer 0xfff0 0x6D 0xB6 0x43 0xCF 0x7E 0x8F 0x47 0x11 0x88 0x66 0x59 0x38 0xD1 0x7A 0xAA 0x26 0x49 0x5E 0x13 0x14 0x15 0x16 0x17 0x18
back
advertise on
`

Hmmm...

elclaudio, 2024-05-14:
> Hi @elclaudio :) I'm absolutely no BT-expert and I also did not manage to send the correct advertisement telegrams with bluetoothctl on my raspi.
>
> Btw.: the manufacturer Id is 0xFFF0
>
> ``` [...]

So, this works in android each time (tested on a old phone and a recent tablet) with nrf connect but never works on linux, be it on x86 or arm. The common stuff is the bt stack, bluez on linux. I still haven't found what is going on. A test on windows could be possible if proper tool or code is available.

J0EK3R, 2024-05-15:
Heureka! :)

I managed to send the proper telegrams with hcitool on my raspi - without knowing what I was doing...

```
sudo su

hciconfig hci0 up
hciconfig hci0 leadv 3

// switch MK6.0 to BT-Mode
hcitool -i hci0 cmd 08 0008 25 02 01 02 1b ff f0 ff 6D B6 43 CF 7E 8F 47 11 88 66 59 38 D1 7A AA 26 49 5E 13 14 15 16 17 18

// stop all channels
hcitool -i hci0 cmd 08 0008 25 02 01 02 1b ff f0 ff 6D B6 43 CF 7E 8F 47 11 84 66 59 38 D1 7A AA 34 67 4A 55 BF 15 16 17 18

// channel 1 fullspeed
hcitool -i hci0 cmd 08 0008 25 02 01 02 1b ff f0 ff 6d b6 43 cf 7e 8f 47 11 84 66 59 47 d1 7a aa 34 67 4a ed b7 15 16 17 18
```

[image]

```
hcitool -i hci0 cmd 08 0008 
length:     25 (hex!)
flags:      02 01 02
length:     1b
type:       ff (-> manufacturer specific)
company id: ff f0
data:       rest
```

J0EK3R, 2024-05-15:
[Here is a link how to increase BLE advertisement](https://stackoverflow.com/questions/21124993/is-there-a-way-to-increase-ble-advertisement-frequency-in-bluez)

elclaudio, 2024-05-15:
> Heureka! :)
> 
> I managed to send the proper telegrams with hcitool on my raspi - without knowing what I was doing...
> 
> ```
> sudo su
> 
> hciconfig hci0 up
> hciconfig hci0 leadv 3
> 
> // switch MK6.0 to BT-Mode
> hcitool -i hci0 cmd 08 0008 25 02 01 02 1b ff f0 ff 6D B6 43 CF 7E 8F 47 11 88 66 59 38 D1 7A AA 26 49 5E 13 14 15 16 17 18
> 
> // stop all channels
> hcitool -i hci0 cmd 08 0008 25 02 01 02 1b ff f0 ff 6D B6 43 CF 7E 8F 47 11 84 66 59 38 D1 7A AA 34 67 4A 55 BF 15 16 17 18
> 
> // channel 1 fullspeed
> hcitool -i hci0 cmd 08 0008 25 02 01 02 1b ff f0 ff 6d b6 43 cf 7e 8f 47 11 84 66 59 47 d1 7a aa 34 67 4a ed b7 15 16 17 18
> ```
> 
> ![Screenshot 2024-05-15 072432](https://private-user-images.githubusercontent.com/24459441/330670765-248d6f27-7d88-4af3-8ba6-c4e82324d84a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3ODgzODIsIm5iZiI6MTcxNTc4ODA4MiwicGF0aCI6Ii8yNDQ1OTQ0MS8zMzA2NzA3NjUtMjQ4ZDZmMjctN2Q4OC00YWYzLThiYTYtYzRlODIzMjRkODRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE1VDE1NDgwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgwN2E3YjBiMjVjNzQ3NGQ1ZjE3MGZkNmIzMTRkMTg4YzQ4ZGI0MTUyODIwYTQyMzFkZDYyZWQxOTVlZjQxM2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.z55bm4FBfJuk2aRZSp4Xvuw0IM00Ows2BDVAUaOz66M)
> 
> ```
> hcitool -i hci0 cmd 08 0008 
> length:     25 (hex!)
> flags:      02 01 02
> length:     1b
> type:       ff (-> manufacturer specific)
> company id: ff f0
> data:       rest
> ```

Fantastic, you are a genius ! ;-) I confirm it work on my raspi zero w too ! 

when launching cmd for full speed, the motor run but make a stop at intervalls and go back to full speed. Do you believe it's related to advertisement speed ? all in all, this is a breakthrough, now a computer can drive any motor from a shell script or programming language !

J0EK3R, 2024-05-15:
> when launching cmd for full speed, the motor run but make a stop at intervalls and go back to full speed. Do you believe it's related to advertisement speed ?
Yes, we have to increase the advertisement interval.

I haven't tested IT but i think [Here is a link how to increase BLE advertisement](https://stackoverflow.com/questions/21124993/is-there-a-way-to-increase-ble-advertisement-frequency-in-bluez) is the solution.

elclaudio, 2024-05-15:
> > when launching cmd for full speed, the motor run but make a stop at intervalls and go back to full speed. Do you believe it's related to advertisement frequency?
> 
> Yes, we have to increase the advertisement speed.
> 
> I haven't tested IT but i think [Here is a link how to increase BLE advertisement](https://stackoverflow.com/questions/21124993/is-there-a-way-to-increase-ble-advertisement-frequency-in-bluez) is the solution.

Yep, instead of 
`hciconfig hci0 leadv 3`

I used this:
```

hcitool -i hci0 cmd 08 0008 25 02 01 02 1b ff f0 ff 6D B6 43 CF 7E 8F 47 11 88 66 59 38 D1 7A AA 26 49 5E 13 14 15 16 17 18
hcitool -i hci0 cmd 0x08 0x0006 A0 00 A0 00 03 00 00 00 00 00 00 00 00 07 00
hcitool -i hci0 cmd 0x08 0x000a 01
```
the motor now run without pause ;-)

J0EK3R, 2024-05-16:
OK, that's fine :)
Now you have to transfer the crypt-function from [here](https://github.com/J0EK3R/brickcontroller2/blob/master/BrickController2/BrickController2/DeviceManagement/MouldKingCrypt.cs) to i.E. python (the function TryDecryptBruteForce) is not needed. All functions in this file are static - that's pretty easy to transfer - no OOP (object oriented programming).
Then, to test, if you put the byte-arrays from [MouldKing_6_0_Modul.cs](https://github.com/J0EK3R/brickcontroller2/blob/master/BrickController2/BrickController2/DeviceManagement/MouldKing_6.0_Modul.cs) in the crypt-function you have to get back the values in my comments.
I think it would make sense to create a git repo for that so we can exchange code and knowhow ;)
Could you create one?

elclaudio, 2024-05-16:
> OK, that's fine :) Now you have to transfer the crypt-function from [...]

Please, excuse me, but I'm no programmer ;-(

However, I tried to convert code from C# to python with online tools, for now I came up with this, the code has no errors but the results arent what they should be :

```
import array
from pprint import pprint

# Define a byte array for Telegram Connect
telegram_connect = [0x6D, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x92]

telegram_base_device_a = [0x61, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9E]
telegram_base_device_b = [0x62, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9D]
telegram_base_device_c = [0x63, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9C]

Array_C1C2C3C4C5 = array.array('B', [0xC1, 0xC2, 0xC3, 0xC4, 0xC5])
MagicArray_37 = [0] * 37
MagicArray_63 = [0] * 63

def Crypt(rawDataArray):
    targetArrayLength = len(Array_C1C2C3C4C5) + len(rawDataArray) + 20

    targetArray = bytearray(targetArrayLength)
    targetArray[15] = 113
    targetArray[16] = 15
    targetArray[17] = 85

    for index in range(len(Array_C1C2C3C4C5)):
        targetArray[index + 18] = Array_C1C2C3C4C5[(len(Array_C1C2C3C4C5) - index) - 1]

    targetArray[18 + len(Array_C1C2C3C4C5):] = rawDataArray

    for index in range(15, len(Array_C1C2C3C4C5) + 18):
        targetArray[index] = revert_bits_byte(targetArray[index])

    checksum = calc_checksum_from_arrays(Array_C1C2C3C4C5, rawDataArray)
    targetArray[len(Array_C1C2C3C4C5) + 18 + len(rawDataArray):] = [(checksum & 255), ((checksum >> 8) & 255)]

    magicNumberArray_63 = MagicArray_63.copy()
    tempArrayLength = targetArrayLength - 18
    tempArray = targetArray[18:].copy()

    crypt_array(tempArray, magicNumberArray_63)
    targetArray[18:] = tempArray

    magicNumberArray_37 = MagicArray_37.copy()
    crypt_array(targetArray, magicNumberArray_37)

    telegramArray = bytearray(24)

    lengthResultArray = len(Array_C1C2C3C4C5) + len(rawDataArray) + 5
    telegramArray[:lengthResultArray] = targetArray[15:15 + lengthResultArray]

    for index in range(lengthResultArray, len(telegramArray)):
        telegramArray[index] = index + 1

    return telegramArray

def create_magic_array(magic_number, size):
    magic_array = [0] * size
    magic_array[0] = 1

    for index in range(1, 7):
        magic_array[index] = (magic_number >> (6 - index)) & 1

    return magic_array

def revert_bits_int(value):
    result = 0
    for index_bit in range(16):
        if ((1 << index_bit) & value) != 0:
            result |= 1 << (15 - index_bit)
    return 65535 & result

def crypt_array(byte_array, magic_number_array):
    # foreach byte of array
    for index_byte in range(len(byte_array)):
        current_byte = byte_array[index_byte]
        current_result = 0
        # foreach bit in byte
        for index_bit in range(8):
            current_result += (((current_byte >> index_bit) & 1) ^ shift_magic_array(magic_number_array)) << index_bit
        byte_array[index_byte] = current_result & 255
    return byte_array

def calc_checksum_from_arrays(first_array, second_array):
    result = 65535
    for first_array_index in range(len(first_array)):
        result = (result ^ (first_array[(len(first_array) - 1) - first_array_index] << 8)) & 65535
        for index_bit in range(8):
            current_result = result & 32768
            result <<= 1
            if current_result != 0:
                result ^= 4129

    for current_byte in second_array:
        result = ((revert_bits_byte(current_byte) << 8) ^ result) & 65535
        for index_bit in range(8):
            current_result = result & 32768
            result <<= 1
            if current_result != 0:
                result ^= 4129

    return revert_bits_int(result) ^ 65535

def shift_magic_array(i_arr):
    # i_arr[3] = i_arr[2]
    # i_arr[2] = i_arr[1]
    # i_arr[1] = i_arr[0]
    # i_arr[0] = i_arr[6]
    # i_arr[6] = i_arr[5]
    # i_arr[5] = i_arr[4]
    # i_arr[4] = i_arr[3] ^ i_arr[6]
    # return i_arr[0]

    r0 = 3
    r1 = i_arr[r0]
    r2 = 6
    r3 = i_arr[r2]
    r1 = r1 ^ r3
    r3 = 2
    r4 = i_arr[r3]
    i_arr[r0] = r4
    r0 = 1
    r4 = i_arr[r0]
    i_arr[r3] = r4
    r3 = 0
    r4 = i_arr[r3]
    i_arr[r0] = r4
    r0 = i_arr[r2]
    i_arr[r3] = r0
    r0 = 5
    r4 = i_arr[r0]
    i_arr[r2] = r4
    r2 = 4
    r4 = i_arr[r2]
    i_arr[r0] = r4
    i_arr[r2] = r1
    r6 = i_arr[r3]
    return r6

def revert_bits_byte(value):
    result = 0
    for index_bit in range(8):
        if ((1 << index_bit) & value) != 0:
            result = result | (1 << (7 - index_bit))
    return result

pprint(Crypt(telegram_connect))
```
the results returned : 

```
>>> %Run MouldKingCrypt.py
bytearray(b'\x8e\xf0\xaa\xa3#\xc3C\x83m{\xa7\x80\x80\x80\x80\x92'
          b'\xae\x8a\x13\x14\x15\x16\x17\x18')
```

J0EK3R, 2024-05-16:
...only a quick review - I don't have time till saturday:
MagicArray_37 and MagicArray_63 aren't properly initialized

J0EK3R, 2024-05-17:
Here is a running version of your code
```
#!/usr/bin/python

import array
from pprint import pprint

# Define a byte array for Telegram Connect
telegram_connect = [0x6D, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x92]

telegram_base_device_a = [0x61, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9E]
telegram_base_device_b = [0x62, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9D]
telegram_base_device_c = [0x63, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9C]

Array_C1C2C3C4C5 = [0xC1, 0xC2, 0xC3, 0xC4, 0xC5]

def Crypt(rawDataArray):
    targetArrayLength = len(Array_C1C2C3C4C5) + len(rawDataArray) + 20

    targetArray = bytearray(targetArrayLength)
    targetArray[15] = 113
    targetArray[16] = 15
    targetArray[17] = 85

    for index in range(len(Array_C1C2C3C4C5)):
        targetArray[index + 18] = Array_C1C2C3C4C5[(len(Array_C1C2C3C4C5) - index) - 1]

    targetArray[18 + len(Array_C1C2C3C4C5):] = rawDataArray

    for index in range(15, len(Array_C1C2C3C4C5) + 18):
        targetArray[index] = revert_bits_byte(targetArray[index])

    checksum = calc_checksum_from_arrays(Array_C1C2C3C4C5, rawDataArray)
    targetArray[len(Array_C1C2C3C4C5) + 18 + len(rawDataArray):] = [(checksum & 255), ((checksum >> 8) & 255)]

    magicNumberArray_63 = create_magic_array(63, 7)
    tempArrayLength = targetArrayLength - 18
    tempArray = targetArray[18:].copy()

    crypt_array(tempArray, magicNumberArray_63)
    targetArray[18:] = tempArray

    magicNumberArray_37 = create_magic_array(37, 7)
    crypt_array(targetArray, magicNumberArray_37)

    telegramArray = bytearray(24)

    lengthResultArray = len(Array_C1C2C3C4C5) + len(rawDataArray) + 5
    telegramArray[:lengthResultArray] = targetArray[15:15 + lengthResultArray]

    for index in range(lengthResultArray, len(telegramArray)):
        telegramArray[index] = index + 1

    return telegramArray

def create_magic_array(magic_number, size):
    magic_array = [0] * size
    magic_array[0] = 1

    for index in range(1, 7):
        magic_array[index] = (magic_number >> (6 - index)) & 1

    return magic_array

def revert_bits_int(value):
    result = 0
    for index_bit in range(16):
        if ((1 << index_bit) & value) != 0:
            result |= 1 << (15 - index_bit)
    return 65535 & result

def crypt_array(byte_array, magic_number_array):
    # foreach byte of array
    for index_byte in range(len(byte_array)):
        current_byte = byte_array[index_byte]
        current_result = 0
        # foreach bit in byte
        for index_bit in range(8):
            current_result += (((current_byte >> index_bit) & 1) ^ shift_magic_array(magic_number_array)) << index_bit
        byte_array[index_byte] = current_result & 255
    return byte_array

def calc_checksum_from_arrays(first_array, second_array):
    result = 65535
    for first_array_index in range(len(first_array)):
        result = (result ^ (first_array[(len(first_array) - 1) - first_array_index] << 8)) & 65535
        for index_bit in range(8):
            current_result = result & 32768
            result <<= 1
            if current_result != 0:
                result ^= 4129

    for current_byte in second_array:
        result = ((revert_bits_byte(current_byte) << 8) ^ result) & 65535
        for index_bit in range(8):
            current_result = result & 32768
            result <<= 1
            if current_result != 0:
                result ^= 4129

    return revert_bits_int(result) ^ 65535

def shift_magic_array(i_arr):
    # i_arr[3] = i_arr[2]
    # i_arr[2] = i_arr[1]
    # i_arr[1] = i_arr[0]
    # i_arr[0] = i_arr[6]
    # i_arr[6] = i_arr[5]
    # i_arr[5] = i_arr[4]
    # i_arr[4] = i_arr[3] ^ i_arr[6]
    # return i_arr[0]

    r0 = 3
    r1 = i_arr[r0]
    r2 = 6
    r3 = i_arr[r2]
    r1 = r1 ^ r3
    r3 = 2
    r4 = i_arr[r3]
    i_arr[r0] = r4
    r0 = 1
    r4 = i_arr[r0]
    i_arr[r3] = r4
    r3 = 0
    r4 = i_arr[r3]
    i_arr[r0] = r4
    r0 = i_arr[r2]
    i_arr[r3] = r0
    r0 = 5
    r4 = i_arr[r0]
    i_arr[r2] = r4
    r2 = 4
    r4 = i_arr[r2]
    i_arr[r0] = r4
    i_arr[r2] = r1
    r6 = i_arr[r3]
    return r6

def revert_bits_byte(value):
    result = 0
    for index_bit in range(8):
        if ((1 << index_bit) & value) != 0:
            result = result | (1 << (7 - index_bit))
    return result

# pprint(Crypt(telegram_connect))
result = Crypt(telegram_connect)
print(' '.join(f'{x:02x}' for x in result))
```
Output:
```
/home/pi/dev/mkconnect/test.py 
6d b6 43 cf 7e 8f 47 11 88 66 59 38 d1 7a aa 26 49 5e 13 14 15 16 17 18
```

elclaudio, 2024-05-17:
I've corrected the code, now it work

```
import array
#from pprint import pprint
#import numpy as np

#np.set_printoptions(formatter={'int':hex})

# Define a byte array for Telegram Connect
telegram_connect = [0x6D, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x92]
telegram_base_device_a = [0x61, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9E]
telegram_base_device_b = [0x62, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9D]
telegram_base_device_c = [0x63, 0x7B, 0xA7, 0x80, 0x80, 0x80, 0x80, 0x80, 0x80, 0x9C]

Array_C1C2C3C4C5 = array.array('B', [0xC1, 0xC2, 0xC3, 0xC4, 0xC5])

def Crypt(rawDataArray):
    targetArrayLength = len(Array_C1C2C3C4C5) + len(rawDataArray) + 20

    targetArray = [0] * targetArrayLength
    targetArray[15] = 113 #0x71
    targetArray[16] = 15 #0x0F
    targetArray[17] = 85 #0x55

    for index in range(len(Array_C1C2C3C4C5)):
        targetArray[index + 18] = Array_C1C2C3C4C5[(len(Array_C1C2C3C4C5) - index) - 1]

    targetArray[18 + len(Array_C1C2C3C4C5):] = rawDataArray

    for index in range(15, len(Array_C1C2C3C4C5) + 18):
        targetArray[index] = revert_bits_byte(targetArray[index])

    checksum = calc_checksum_from_arrays(Array_C1C2C3C4C5, rawDataArray)
    targetArray[len(Array_C1C2C3C4C5) + 18 + len(rawDataArray):] = [(checksum & 255), ((checksum >> 8) & 255)]

    magicNumberArray_63 = create_magic_array(63, 7)
    tempArrayLength = targetArrayLength - 18
    tempArray = targetArray[18:].copy()

    crypt_array(tempArray, magicNumberArray_63)
    targetArray[18:] = tempArray

    magicNumberArray_37 = create_magic_array(37, 7)
    crypt_array(targetArray, magicNumberArray_37)

    telegramArray = [0] * 24

    lengthResultArray = len(Array_C1C2C3C4C5) + len(rawDataArray) + 5
    telegramArray[:lengthResultArray] = targetArray[15:15 + lengthResultArray]

    for index in range(lengthResultArray, len(telegramArray)):
        telegramArray[index] = index + 1

    return telegramArray

def create_magic_array(magic_number, size):
    magic_array = [0] * size
    magic_array[0] = 1

    for index in range(1, 7):
        magic_array[index] = (magic_number >> (6 - index)) & 1

    return magic_array

def revert_bits_int(value):
    result = 0
    for index_bit in range(16):
        if ((1 << index_bit) & value) != 0:
            result |= 1 << (15 - index_bit)
    return 65535 & result

def crypt_array(byte_array, magic_number_array):
    # foreach byte of array
    for index_byte in range(len(byte_array)):
        current_byte = byte_array[index_byte]
        current_result = 0
        # foreach bit in byte
        for index_bit in range(8):
            current_result += (((current_byte >> index_bit) & 1) ^ shift_magic_array(magic_number_array)) << index_bit
        byte_array[index_byte] = current_result & 255
    return byte_array

def calc_checksum_from_arrays(first_array, second_array):
    result = 65535
    for first_array_index in range(len(first_array)):
        result = (result ^ (first_array[(len(first_array) - 1) - first_array_index] << 8)) & 65535
        for index_bit in range(8):
            current_result = result & 32768
            result <<= 1
            if current_result != 0:
                result ^= 4129

    for current_byte in second_array:
        result = ((revert_bits_byte(current_byte) << 8) ^ result) & 65535
        for index_bit in range(8):
            current_result = result & 32768
            result <<= 1
            if current_result != 0:
                result ^= 4129

    return revert_bits_int(result) ^ 65535

def shift_magic_array(i_arr):
    # i_arr[3] = i_arr[2]
    # i_arr[2] = i_arr[1]
    # i_arr[1] = i_arr[0]
    # i_arr[0] = i_arr[6]
    # i_arr[6] = i_arr[5]
    # i_arr[5] = i_arr[4]
    # i_arr[4] = i_arr[3] ^ i_arr[6]
    # return i_arr[0]

    r0 = 3
    r1 = i_arr[r0]
    r2 = 6
    r3 = i_arr[r2]
    r1 = r1 ^ r3
    r3 = 2
    r4 = i_arr[r3]
    i_arr[r0] = r4
    r0 = 1
    r4 = i_arr[r0]
    i_arr[r3] = r4
    r3 = 0
    r4 = i_arr[r3]
    i_arr[r0] = r4
    r0 = i_arr[r2]
    i_arr[r3] = r0
    r0 = 5
    r4 = i_arr[r0]
    i_arr[r2] = r4
    r2 = 4
    r4 = i_arr[r2]
    i_arr[r0] = r4
    i_arr[r2] = r1
    r6 = i_arr[r3]
    return r6

def revert_bits_byte(value):
    result = 0
    for index_bit in range(8):
        if ((1 << index_bit) & value) != 0:
            result = result | (1 << (7 - index_bit))
    return result

print(Crypt(telegram_connect))
print(Crypt(telegram_base_device_a))
```

elclaudio, 2024-05-17:
> Here is a running version of your code
>
> ```
> #!/usr/bin/python
>
> import array
> from pprint import pprint
>
> # Define a byte array for Telegram Connect
> telegram_connect = [0x6D, 0x7B, 0xA7, 0x80, 0x80, [...]

Oh thanks, sorry, I was looking at it in beetwen, I didn't see your post !

telegram_base_device_a seem to be the code for stopping the motors but how do you manage the choice of channel, rotation direction and motor's power ?

elclaudio, 2024-05-17:
https://github.com/imurvai/brickcontroller2/assets/6136831/78c71be0-c5bd-4b8c-baab-36d14fdf9dc9

Many things to do yet, but very promising ! I will order a raspberry pi pico W (with embedded python interpreter and bluetooth) to test if that works too... but I have to find a way to bypass hcitool and send the commands directly to the integrated BT controller I hope it's possible on this device, if not I'll keep the raspi zero w. 

Next step is to control channel selection, motor direction and motor power. After that, comes the nice part of using proximity sensor, color sensors and switchs via i2c which should be easy to do. In the end, why not make a pcb board with sensors connectors (i2c bus allow many peripherals on short distances with minimal cabling) and digital input (I think about magnetics dil relays, very cheap, and others simple mecanicals switchs, all those can act as limit switch)

I have another mould king 6.0 hub on order plus a bunch of motors to make tests

J0EK3R, 2024-05-18:
Oh, i found something interesting: 
[hcitool has been deprecated](https://raspberrypi.stackexchange.com/questions/111740/raspberry-pi-bluetooth-hci-event-code-meaning)

J0EK3R, 2024-05-19:
@elclaudio 
I have created the [repo mkconnect-python](https://github.com/J0EK3R/mkconnect-python).
I suggest to continue the discussion there... :)