imurvai/brickcontroller2 #92: Request for Flysky G7P support (works but not all buttons)

BatteryPoweredBricks, 2022-12-24:
Hello, First off thank you for the awesome app!  I was thrilled to find an alternative to using touch controls.  Using a PS4 controller didn't feel quite right so I tried a few options with real RC controllers.  My Flysky G7P that arrived today has a USB-C port and functions as a game controller.  It works fine in the app for throttle and steering (which is a great first step!), but the app does not see the other buttons being pressed.  I tried another android app and the buttons do show up there.  If I can provide data for the button addresses can they be added to the BrickController app?  I'm no coder myself but will help in any way I can.  This is a huge milestone for Lego RC to be able to use a pistol grip controller without any modifications.  It would be that much better to be able to control locking differentials, gears, lights etc. from the same controller.

Thank you for your time!
Brian - BatteryPoweredBricks

BatteryPoweredBricks, 2022-12-24:
[image]
I was able to get this much information from Gamepad Tester.  From my limited understanding the Scancodes / Key Codes are different from what a typical controller would have.  I've looked into remapping the buttons in the mean time and haven't come up with much.

hiryu2, 2025-06-08:
By assigning a button to the channel in the setting of the transmitter, it can be controlled by the Brick Controller2 app.