imurvai/brickcontroller2 #94: Build errors in multidex

Oliver-Habersetzer, 2023-01-11:
Hello!

I've been trying to build the android version of the app inside VS22 but I encountered a issue with multidex:

Build log:
```log
Build started...
Starting emulator pixel_5_-_api_31 ...
C:\Program Files (x86)\Android\android-sdk\emulator\emulator.EXE -netfast -avd pixel_5_-_api_31 -prop monodroid.avdname=pixel_5_-_api_31
1>------ Build started: Project: BrickController2.Android, Configuration: Debug Any CPU ------
1>C:\coding\brickcontroller2\BrickController2\BrickController2\BrickController2.csproj : warning NU1701: Package 'SQLiteNetExtensions 2.0.0' was restored using '.NETFramework,Version=v4.6.1, .NETFramework,Version=v4.6.2, .NETFramework,Version=v4.7, .NETFramework,Version=v4.7.1, .NETFramework,Version=v4.7.2, .NETFramework,Version=v4.8, .NETFramework,Version=v4.8.1' instead of the project target framework '.NETStandard,Version=v2.1'. This package may not be fully compatible with your project.
1>  BrickController2.Android -> C:\coding\brickcontroller2\BrickController2\BrickController2.Android\bin\Debug\BrickController2.Android.dll
1>D8 : error : Cannot fit requested classes in a single dex file (# methods: 67347 > 65536). Try supplying a main-dex list
1>MSBUILD : java.exe error JAVA0000: Warning in obj\Debug\120\lp\92\jl\classes.jar:androidx/activity/OnBackPressedDispatcher$Api33Impl.class:
1>MSBUILD : java.exe error JAVA0000: Type `android.window.OnBackInvokedCallback` was not found, it is required for default or static interface methods desugaring of `androidx.activity.OnBackPressedDispatcher$Api33Impl$$InternalSyntheticLambda$2$f0f438d88c205875ff38e25fbc26be78a4fda65d6bee8c4d41a08d272c7d9f6e$0`
1>MSBUILD : java.exe error JAVA0000: Warning in obj\Debug\120\androidx\jl\00bcb56b-b993-46a0-8199-052eaf941e3c.jar:kotlinx/coroutines/internal/ClassValueCtorCache$cache$1.class:
1>MSBUILD : java.exe error JAVA0000: Type `java.lang.ClassValue` was not found, it is required for default or static interface methods desugaring of `kotlinx.coroutines.internal.ClassValueCtorCache$cache$1`
1>MSBUILD : java.exe error JAVA0000: Warning in obj\Debug\120\androidx\jl\00bcb56b-b993-46a0-8199-052eaf941e3c.jar:kotlinx/coroutines/debug/AgentPremain$DebugProbesTransformer.class:
1>MSBUILD : java.exe error JAVA0000: Type `java.lang.instrument.ClassFileTransformer` was not found, it is required for default or static interface methods desugaring of `kotlinx.coroutines.debug.AgentPremain$DebugProbesTransformer`
1>MSBUILD : java.exe error JAVA0000: Warning in obj\Debug\120\androidx\jl\00bcb56b-b993-46a0-8199-052eaf941e3c.jar:kotlinx/coroutines/debug/AgentPremain.class:
1>MSBUILD : java.exe error JAVA0000: Type `sun.misc.SignalHandler` was not found, it is required for default or static interface methods desugaring of `kotlinx.coroutines.debug.AgentPremain$$InternalSyntheticLambda$2$7895cd395e43c061a299e224a1d3672f97bd4610fe97f0e188c9c199a1620b54$0`
1>MSBUILD : java.exe error JAVA0000: Error: Cannot fit requested classes in a single dex file (# methods: 67347 > 65536). Try supplying a main-dex list
1>MSBUILD : java.exe error JAVA0000: Compilation failed
1>MSBUILD : java.exe error JAVA0000: java.lang.RuntimeException: com.android.tools.r8.CompilationFailedException: Compilation failed to complete, position
1>MSBUILD : java.exe error JAVA0000:  null
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.Fj.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:98)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.D8.main(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:4)
1>MSBUILD : java.exe error JAVA0000: Caused by: com.android.tools.r8.CompilationFailedException: Compilation failed to complete, position: null
1>MSBUILD : java.exe error JAVA0000: 	at Version.fakeStackEntry(Version_3.3.75.java:0)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.Fj.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:75)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.Fj.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:28)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.Fj.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:27)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.Fj.b(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:2)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.D8.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:22)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.D8.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:17)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.Fj.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:85)
1>MSBUILD : java.exe error JAVA0000: 	... 1 more
1>MSBUILD : java.exe error JAVA0000: Caused by: com.android.tools.r8.internal.f: Cannot fit requested classes in a single dex file (# methods: 67347 > 65536)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.JT.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:14)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.JT.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:22)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.n30.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:524)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.i30.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:7)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.F2.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:371)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.F2.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:455)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.D8.d(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:756)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.D8.c(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:1)
1>MSBUILD : java.exe error JAVA0000: 	at com.android.tools.r8.internal.Fj.a(R8_3.3.75_b7a6ff6b13548611571508fe72282c9167faa649161ca0013edfc92e19bd7e58:24)
1>MSBUILD : java.exe error JAVA0000: 	... 5 more
1>MSBUILD : java.exe error JAVA0000: Directory 'obj\Debug\120\lp\92' is from 'androidx.activity.activity.aar'.
========== Build: 0 succeeded, 1 failed, 1 up-to-date, 0 skipped ==========
========== Elapsed 00:30,038 ==========
========== Deploy: 0 succeeded, 0 failed, 0 skipped ==========
========== Elapsed 00:30,038 ==========
Emulator pixel_5_-_api_31 is running.
```

Error list contents:
```error
Severity	Code	Description	Project	File	Line	Suppression State
Error		Warning in obj\Debug\120\lp\92\jl\classes.jar:androidx/activity/OnBackPressedDispatcher$Api33Impl.class:
Type `android.window.OnBackInvokedCallback` was not found, it is required for default or static interface methods desugaring of `androidx.activity.OnBackPressedDispatcher$Api33Impl$$InternalSyntheticLambda$2$f0f438d88c205875ff38e25fbc26be78a4fda65d6bee8c4d41a08d272c7d9f6e$0`
Warning in obj\Debug\120\androidx\jl\00bcb56b-b993-46a0-8199-052eaf941e3c.jar:kotlinx/coroutines/internal/ClassValueCtorCache$cache$1.class:
Type `java.lang.ClassValue` was not found, it is required for default or static interface methods desugaring of `kotlinx.coroutines.internal.ClassValueCtorCache$cache$1`
Warning in obj\Debug\120\androidx\jl\00bcb56b-b993-46a0-8199-052eaf941e3c.jar:kotlinx/coroutines/debug/AgentPremain$DebugProbesTransformer.class:
Type `java.lang.instrument.ClassFileTransformer` was not found, it is required for default or static interface methods desugaring of `kotlinx.coroutines.debug.AgentPremain$DebugProbesTransformer`
Warning in obj\Debug\120\androidx\jl\00bcb56b-b993-46a0-8199-052eaf941e3c.jar:kotlinx/coroutines/debug/AgentPremain.class:
Type `sun.misc.SignalHandler` was not found, it is required for default or static interface methods desugaring of `kotlinx.coroutines.debug.AgentPremain$$InternalSyntheticLambda$2$7895cd395e43c061a299e224a1d3672f97bd4610fe97f0e188c9c199a1620b54$0`
[... cut ...]
Also I noticed that there is no Solution Platform especially for Android although there are options for iOS:
[image]

This isn't critical - but myb one of you encountered a similar issue or you have an idea what I might have misconfigured?

Googling that issue only gave me this: https://stackoverflow.com/questions/48249633/errorcannot-fit-requested-classes-in-a-single-dex-file-try-supplying-a-main-dex

The solution seems to not apply though since multidex is enabled inside the repo:
[image]

Have a good day :)

Oliver-Habersetzer, 2023-01-11:
Small addition: It seems that my hardware device isn't supported? I'm running the Android 13 on my Samsung S20 FE
[image]

imurvai, 2023-01-12:
Hi, the multidex issue has been fixed in the Release 3.3 branch but it hasn't been merged back to master yet. There is an option in the Android project settings for it though:
[image]

imurvai, 2023-01-12:
Your device should be supported and the playstore says all of your devices are supported actually. Not sure what the issue might be.

Oliver-Habersetzer, 2023-01-17:
> Hi, the multidex issue has been fixed in the Release 3.3 branch but it hasn't been merged back to master yet. There is an option in the Android project settings for it though: [image]

Ahhh yes it builds now!
It doesn't open in the emulator but I'll have to try on my phone first. myb it's just my config here. This is resolved now thouh
Thanks a lot, this project is great! <3

closed 2023-01-17 by Oliver-Habersetzer