imurvai/brickcontroller2 #95: port number left and right is swapped on Buwizz2.0

TianchengTang, 2023-01-19:
Hi,

I have a BuWizz 2.0 Ludicrous and brick controller 2 installed on iphone. Somehow the 1/2 and 3/4 port are swapped while controlling the motor. To elaborate, when I set to start on port 1, the motor on port 2 is spinning. When I set to start on port 3, port 4 is spinning and vice versa. Looks like mapping is flipped left and right. I have double checked port number on Buwizz. Maybe worth debugging. Thank you for this great app!

Regards,
Kingsley

imurvai, 2023-01-20:
Hi,
Yes, it's a known issue that for some BW2 devices the ports are swapped. Unfortunately I could not figure out a patter yet how to identify the affected BW2s.
Sorry for the incovenience.
BR

vicocz, 2023-01-20:
> Does it make sense to add per device setting for channel Buwizz2 swapping instead of detecion based on the firmware revision? please? At least my B2 has `0x1E` and therefore it swaps channels which is wrong in my case.

_Originally discussed by @vicocz in https://github.com/imurvai/brickcontroller2/issues/20#issuecomment-995010901_

vicocz, 2024-12-26:
I've added settings for BuWizz2 so as it can be overriden (via new Device Settings page). See https://github.com/vicocz/brickcontroller2/pull/64/files
If there is will, I can backport it here as well.