imurvai/brickcontroller2 #96: Support for joystick mixing to get arcade style tankcontrol.

gearsincorg, 2023-03-23:
Hi

I'm looking to use BC2 for LEGO Combat robots.  These robots typically have 2 wheel differential drives, and a thrid weapon motor.
I'd like to be able to use a single XY joystick to run the two wheel differential drive,
eg:  X would steer and Y would drive (rather than simple tank drive that takes two thumbs)

Is there current/future support available/planned for this type of control mode?

I'm a Java programmer who uses Android Studio, so I'm thinking it would be difficult for me to try adding it myself.

Thanks
Phil.

jkavalik, 2023-04-09:
@gearsincorg With my son we built the `42140 - App-Controlled Transformation Vehicle` today and I was trying different configurations. In the end I found out that it is possible:
You map Y to both A and B (inversed here)
Then you map X again to both A and B (but no inversion)

That way moving the stick only up-down drives forward and backward and moving it only left-right turns the vehicle in place counter/clockwise.
The way the internals work it sums input from both axes to the same motor, then clamps the results to <-1;1>, so even under full throttle when you move the stick to turn, one of the engines stays at 1 and the other one gets a negative valued added to the full throtle and slows down that motor.
I tested it with my ps4 controller and was able to do it. On the other side it is a bit more responsive to have throttle on one stick and the steering on the other, but that is because how the sticks are working on that controller (and why most racing games use the triggers as gas/brake probably).

gearsincorg, 2023-04-10:
Thanks for the great information.   
 It didn't occur to me that multiple inputs would be additive like that.  I assumed a 1:1 correlation between sticks and motors.

jkavalik, 2023-04-10:
I actually wanted to try and implement it, checking out the Windows-compatible fork to try some changes and I found out in the process how multiple inputs are combined and that no change is needed :)