imurvai/brickcontroller2 #97: Another Logitech F710 question, it works but..

Rockstede, 2023-04-01:
I can't add any controller events to a profile because the app is expecting a Bluetooth controller to be connected instead of a usb>adapter>phone connected controller, is there anyway around this?

Logitech F710 gamepad
Samsung Note 9
USB adapter + F710 receiver

imurvai, 2023-04-01:
Hi,
Actually it's up to the Android how it handles the USB devices. The app is just processing the standard controller events coming from the Android OS. So please look for a driver or something for your controller and Android.
BR,
István

Rockstede, 2023-04-01:
Hmm, but if the app already sees the correct inputs doesn't that imply that a driver isn't required? Or is already present?