imurvai/brickcontroller2 #110: CADA hub support

madbartje, 2024-01-01:
Is it an idea to add the support for CADA hub? Many thanks,
Bart (netherlands)

J0EK3R, 2024-04-03:
Hi Bart,

in my fork of this repo, I have implemented support for the CADA C51072W Car...

Sorry, android only: [Release J0EK3R3.3](https://github.com/J0EK3R/brickcontroller2/releases/tag/J0EK3R3.3)

MiyouQ, 2024-07-21:
> Hi Bart,
> 
> in my fork of this repo, I have implemented support for the CADA C51072W Car, the red one only...
> 
> Sorry, android only: [Release J0EK3R3.3](https://github.com/J0EK3R/brickcontroller2/releases/tag/J0EK3R3.3)

Great work! Will there be plans to support mouldking 13130 black batteries in the future?

J0EK3R, 2024-07-21:
Hi MiyouQ!
Thank you for the flowers ;)
Did you test the black diy hubs with the already implemented MK4.0 stuff?

MiyouQ, 2024-07-21:
Yes, I've tested it, but it doesn't work.

J0EK3R, 2024-07-21:
I've added an issue [add support mouldking 13130 black diy module](https://github.com/J0EK3R/brickcontroller2/issues/2)

J0EK3R, 2024-08-11:
I just added the support for the black MouldKing DIY Module...

[Release 3.5 J0EK3R Beta 0](https://github.com/J0EK3R/brickcontroller2/releases/tag/3.5_J0EK3R_Beta_0)

GusFantasma, 2024-09-03:
Hi J0EKRR, could you share how to link correctly the hub, it seem it does not work.

J0EK3R, 2024-09-03:
Hi @GusFantasma 

> Hi J0EKRR, could you share how to link correctly the hub, it seem it does not work.

Yes, it should not be very complicated... ;)

Install the latest release of *my build of the brickcontroller app* - sorry, Android only, press the button on the hub and the button (magnifying glas) in the app and the *Mould King DIY* should appear.

Here is a link to the latest release:
[3.5_J0EK3R_Beta_1](https://github.com/J0EK3R/brickcontroller2/releases/tag/3.5_J0EK3R_Beta_1)

Or did you mean the CADA Hub...?
It's not supported yet...

GusFantasma, 2024-09-04:
> Hi @GusFantasma
>
>
> Yes, it should not be very complicated... ;)
>
> Install the latest release of _my build of the brickcontroller app_ - sorry, Android only, press the button on the hub and the button [...]

Thanks, I'm working with the CaDA C51072W Race Car but it doesn't pair.

J0EK3R, 2024-09-04:
@GusFantasma 
I added an [issue](https://github.com/J0EK3R/brickcontroller2/issues/3) in my repo.

J0EK3R, 2024-10-06:
I think I've solved the 10000 pieces puzzle of a dark sky...

[4.0 J0EK3R Beta 1](https://github.com/J0EK3R/brickcontroller2/releases/tag/4.0_J0EK3R_Beta_1)

I can control my two CaDA RaceCars by the app - at the same time! :)

Just press the scan button and power on your CaDA RaceCars - they should be found by the app and appear in the devices list.

---
Btw: I did not found any *CaDA hubs* wich can be controlled by *Bluetooth*.
I think there are only RC controlled hubs...

GusFantasma, 2024-10-07:
Thank you so much, now it works! I just need a new joypad to have better control 😃

J0EK3R, 2024-10-07:
You're welcome! :)