imurvai/brickcontroller2 #114: Add release images

muellerj, 2024-06-01:
Hi there,

first of all thanks a lot for all the work you put into this project. I'm on iOS trying to control a CircuitCube - the version of the app currently in the App Store doesn't seem to support it (Version 3.1). I tried building the project myself, but VSCode for Mac seems to be discontinued so I can't get the Xamarin framework running.

Is there a possibility of adding the latest images as releases on Github? Or possibly update the version on the App Store?

Kind regards -
Jonas

imurvai, 2024-06-02:
Hi, You need to install Visual Studio for Mac (not VS Code) in order to compile the project. BUT! Recently Xamarin Forms has been discontinued, so I started working on updating the whole project. Anyway maybe it would be worth giving a try to compile.

muellerj, 2024-06-02:
Ah, thanks for the pointer. I'll try that tomorrow - if you need someone to beta test the rewrite let me know. 

I really appreciate what you've built here.

vicocz, 2024-06-19:
@muellerj AFAIK CircuitCube does not seem to be discovered on iOS (I've tracked as in my fork https://github.com/vicocz/brickcontroller2/issues/4).
I Would expect that on IOS there needs to be some setup do propery configure BLE advetisment data (this should relate to discovere code here: https://github.com/vicocz/brickcontroller2/blob/a32422ed99ec1abee2ddee095111e0579fd20e63/BrickController2/BrickController2/DeviceManagement/BluetoothDeviceManager.cs#L109)

muellerj, 2024-06-20:
Thanks so much for your help! I have not (yet) managed to download BC to my phone.

Just out of interest - how does the CircuitCube app (https://apps.apple.com/us/app/circuit-cubes/id1530623322) handle the connection? The app is crap in many other ways including not being able to detect simultaneous button presses properly, so I can't use that either :/