imurvai/brickcontroller2 #118: Add support for BuWizz 3.0 current limiter

BrickExperimentChannel, 2024-08-02:
Hello,

Could you add a current limiter support for BuWizz 3.0?
Without it, Buggy motors sometimes cause the unit to shut down.

The specs is here, at 0x38 Set current limits
https://buwizz.com/BuWizz_3.0_API_3.22_web.pdf

The specs says the payload value is in steps of 30 mA, but I think it is really 35 mA. Their official BuWizz app sends 0x3C for 2.1A and 0x09 for 0.32A.

For it to work, the LUT options must be 1 in command 0x31 Set motor data

You can download my reference implementation here.
https://www.dropbox.com/scl/fo/mr4h8oyet4d2gzv78ra3k/AAPulxX1pFI_cAGAQ5Howok?dl=0&e=1&preview=com.scn.BrickController2_source.zip&rlkey=5nmbh3femf4sto95rvriucnvn

If you can add a way for user to configure the current limit in the UI, that would be extra great. The default limit is 2.1A in the BuWizz app.

Thanks.

imurvai, 2024-08-19:
Thanks for the research, I'll check it.

Avera9eJoe, 2024-08-26:
Commenting here on behalf of those with Combat Robotics Brick League and Battlebots, there's quite a number of us that would greatly appreciate this as well - several matches have ended early when one bot has shut off randomly for what we now suspect is this reason