imurvai/brickcontroller2 #120: Stop movement when USB is detached

BrickExperimentChannel, 2024-08-02:
I'm using your app with a PS4 controller via an USB cable. USB OTG adapter is in between.
Sometimes the cable is detached when I'm driving a boat at full speed and the boat keeps going on. It leads to dangerous situations.

Could the app reset all all controls when the USB cable is detached?

I have done an ugly implementation that works for me but I'm not sure how reliable it is. Catch MotionEventActions.Cancel and then reset proportional controls. Resetting Accelerator-type controls is something I couldn't figure out how to do properly.
https://www.dropbox.com/scl/fo/mr4h8oyet4d2gzv78ra3k/AAPulxX1pFI_cAGAQ5Howok?dl=0&e=1&preview=com.scn.BrickController2_source.zip&rlkey=5nmbh3femf4sto95rvriucnvn

imurvai, 2024-08-19:
hmm, interesting configuration with the OTG cable. Just curious: are you using the cable becase of the lower latency?
What I can imagine is to fire events for the joy events with zero values so that it could be caught by the app then stop the corresponding outputs.

BrickExperimentChannel, 2024-08-20:
Yes, lower latency and better reliability. There are fewer delay spikes with the cable, I think.

This OTG adapter I use.
https://www.swissten.eu/en/usb-splitter-powered/swissten-otg-adapter-lightning-male--usb-a-female/

Zeroing out the joy events works nicely for normal proportional controls, like PS4 sticks. I tested it multiple times. But Accelerator-type controls, which I have configured for D-pad to sustain throttle at a constant level, need something else. They don't listen to joy events the same way. Maybe the Stop-button functionality could be invoked because it resets accelerators?