imurvai/brickcontroller2 #123: Porsche GT4 e-Performance does not connect to BrickController 2

legomania76, 2024-10-09:
Dear imurvai,

I recently got the Lego Technic Porsche GT4 e-Performance wil Control features. Building it was a blast, racing it even more so!

However it wasn't long until the inaccurate controls of my touchscreen started to annoy me. I google searched if there was a way to Control the car with a PS5 controller. It wasn't long until I stumbled upon BrickController2.

Unfortunately the GT4 doesn't seem to be found by the app. Perhaps because Lego used a new type of electro motor? Either way, I stumbled upon this Github thread where a user named toorisrael claims to have come up with the following solution to the code: 

https://github.com/toorisrael/brickcontroller2/commit/5cf3b4f699f3b3ea04e3473d000b5da065f2fce2#diff-87d78588d3370a3340d0382623242875748387d6c1b2ea5d0c09768a549b5f17R99

I am not tech savvy enough myself to implement these changes into the code of the BrickController2 app. Would you please be so kind as to add this commit to the app. That way end users like me can simply download it from the Play Store. Thanks a bundle in advance!

André

vicocz, 2024-10-14:
I spent some time with own prototype for this new hub, but it's not as easy as adding a new type of the device. There seemed to be some incompatibility and existing Control+ command did not worked very well.
I plan to try it again soon, can keep this thread updated.

vicocz, 2024-10-26:
There is DRAFT of my work-in-progress here: https://github.com/imurvai/brickcontroller2/pull/125

imurvai, 2024-10-27:
Hi, that is great that you are working on it. Unfortunately I don't have that set, so I cannot try/test it. Let me know when you think it's ready to be merged and I'll release it.

vicocz, 2024-11-13:
Seems to be working now. It requiered a lot of "hacking" to apply PLAYVM mode which drives both AB motors and C servo. I hope they will fix generic support for LPW3 command soon.
Would be nice if someone could test it as well. But feel free to merge it or leave comments in PR to suggest any change.

valmanassiev, 2024-12-11:
@vicocz I've tried this dev on my sons GT4.
I have checkout vicocz:feature/42176-playvm-mode and build the project for Android.

The main issue that I have is that commands to Ch AB does not seem trigger any action.
I've tried different mappings:
- Ltrigger AB-inv /Rtrigger AB   with  JoyType=Normal/Train/Acceleration
- Y-axis AB
- Map AB on other buttons 

Lights and the channel C are working fine.

I am using PS5 controller on a pixel 7.
I am not familiar with .Net/Maui and Android development. 
I've used conneqthub/dotnet-maui-android:8.0.72 docker image to build the apk

Let me know if I can provide other inputs.

vicocz, 2024-12-12:
Hi, thx for the feedback. The only thing you can provide is GT's firmware revision.
I guess tha gamepad won't be the issue if it works on the test page.
Anyway it's weird, I used to have issues with C channel. 
Btw have you setup the servo for C channel? I guess yes, but could you share the controller profile page please?

valmanassiev, 2024-12-12:
I've switched the C channel to Servo and it start working ;)
Also increased the dead zones for the triggers and axis to 5%-10% just to avoid the jitter.
So now we are having fun with it. 
Thanks, great work!

vicocz, 2024-12-12:
Anyway, I've addressed such edge case, PLAYVM will not be applied and it should somehow be driving, even though it's not ideal to drive C channel without proper servo setup. But it works 😄

vicocz, 2024-12-14:
@imurvai Seems it's working now, feel free to merge it or leave a comment in PR if there is something wrong or to be changed.

imurvai, 2024-12-15:
@vicocz Sure, once I have a little time, I'll do it

jtregl, 2025-01-05:
Hi @vicocz 
I have checkout the **vicocz:feature/42176-playvm-mode** fork, build it for android and install on my mobile phone. But the app is not able to detect the Move Hub device. Is it meant to be detected same way like the Technic Hub or there must be made some other magic in the phone settings or somewhere else?
Thanks for hints in advanced.

jtregl, 2025-01-06:
Hi,
I have tested the **vicocz:feature/42176-playvm-mode**. Everything is working perfectly. 
I just don't know how to setup the handbrake functionality and light up the brake lights from off or 30% intensity to 100% when the button for handbrake is pressed and switch the beack lights back to original state (off or 30%), when the button is released.

vicocz, 2025-01-06:
> Hi, I have tested the **vicocz:feature/42176-playvm-mode**. Everything is working perfectly.

Good to hear, thx for the feedback.

> I just don't know how to setup the handbrake functionality and light up the brake lights from off or 30% intensity to 100% when the button for handbrake is pressed and switch the beack lights back to original state (off or 30%), when the button is released.

May be you can try to somehow combine a sequence and ;Stop; button type:
- for forward-backward command assign AB channel for movement and some sequence with zero value to back lights
- for brake button, you can try to apply the `Stop` button type for AB channel and appply a sequence wih 30 percent value for back lights.

Just an idea, I havn't tested that 😄

wojciechszymik, 2025-01-26:
Hey @imurvai, @vicocz. Great work on the app, it's really useful! 
Any plans soon to merge this PR with support for 42176 and releasing new android app?

vicocz, 2025-01-27:
@wojciechszymik Thank you, I'm just a contributor, so I have no answer related to any release date.
I personally keep my fork with additional features including support for Porsche, you can try the most recent dev build: https://github.com/vicocz/brickcontroller2/actions/runs/12967591255

wojciechszymik, 2025-01-28:
thanks @vicocz ! I tested it yesterday and it works rly nice 🙌

khashashin, 2025-02-13:
Can someone share controller profile for ps5 controller?

I've installed the build generated by GH actions and added Rtrigger for AB but it does not work

khashashin, 2025-02-13:
Stearing works hw

vicocz, 2025-02-13:
Here is mine, copy pasted from JSON and made it pretty:
```
{
    "ct": "bc2c",
    "p": {
        "Name": "Porsche PLAYVM",
        "ControllerProfiles": [
            {
                "Name": "Default profile",
                "ControllerEvents": [
                    {
                        "ControllerId": "",
                        "EventCode": "ButtonY",
                        "ControllerActions": [
                            {
                                "DeviceId": "TechnicMove#CC:45:A5:8A:34:73",
                                "Channel": 8,
                                "ButtonType": 1,
                                "MaxOutputPercent": 100,
                                "AxisActiveZonePercent": 100,
                                "MaxServoAngle": 90,
                                "StepperAngle": 90,
                                "SequenceName": ""
                            },
                            {
                                "DeviceId": "TechnicMove#CC:45:A5:8A:34:73",
                                "Channel": 7,
                                "ButtonType": 1,
                                "MaxOutputPercent": 100,
                                "AxisActiveZonePercent": 100,
                                "MaxServoAngle": 90,
                                "StepperAngle": 90,
                                "SequenceName": ""
                            }
                        ]
                    },
                    {
                        "ControllerId": "",
                        "EventType": 1,
                        "EventCode": "Rz",
                        "ControllerActions": [
                            {
                                "DeviceId": "TechnicMove#CC:45:A5:8A:34:73",
                                "Channel": 12,
                                "IsInvert": true,
                                "MaxOutputPercent": 100,
                                "AxisActiveZonePercent": 100,
                                "MaxServoAngle": 90,
                                "StepperAngle": 90,
                                "SequenceName": ""
                            }
                        ]
                    },
                    {
                        "ControllerId": "",
                        "EventType": 1,
                        "EventCode": "X",
                        "ControllerActions": [
                            {
                                "DeviceId": "TechnicMove#CC:45:A5:8A:34:73",
                                "Channel": 2,
                                "ChannelOutputType": 1,
                                "MaxOutputPercent": 100,
                                "AxisActiveZonePercent": 100,
                                "MaxServoAngle": 90,
                                "StepperAngle": 90,
                                "SequenceName": ""
                            }
                        ]
                    },
                    {
                        "ControllerId": "",
                        "EventCode": "ButtonX",
                        "ControllerActions": [
                            {
                                "DeviceId": "TechnicMove#CC:45:A5:8A:34:73",
                                "Channel": 4,
                                "ButtonType": 1,
                                "MaxOutputPercent": 100,
                                "AxisActiveZonePercent": 100,
                                "MaxServoAngle": 90,
                                "StepperAngle": 90,
                                "SequenceName": ""
                            },
                            {
                                "DeviceId": "TechnicMove#CC:45:A5:8A:34:73",
                                "Channel": 5,
                                "ButtonType": 1,
                                "MaxOutputPercent": 100,
                                "AxisActiveZonePercent": 100,
                                "MaxServoAngle": 90,
                                "StepperAngle": 90,
                                "SequenceName": ""
                            }
                        ]
                    }
                ]
            }
        ]
    }
}
```

Daidalos117, 2025-03-06:
Can we please deploy it ? 🙏

akosyakov, 2025-03-16:
@imurvai Would it be possible to make a release? 🙏 

@vicocz It seems Android build is expired?

I'm new to it. What is the easiest way to try it on the iOS?

sascha024, 2025-05-13:
Guten Tag 

Ist es möglich motor a und motor b unabhängig voneinander zu steuern

Mit freundlichen Grüßen

sascha024, 2025-05-15:
Guten Tag 

Ich habe zuletzt 2025-2 genutzt

Mfg

vicocz, 2025-05-16:
> Ich habe zuletzt 2025-2 genutzt

Hi, yes, that version should support that switch, it set per device, so just open Device settings page and if you switch this setting (called Easy drive mode (PLAYVM)) OFF, there will be both A and B channels available instead:

[image]

There is one serious consequence of that: port C will not support SERVO mode then.

Best Regards