imurvai/brickcontroller2 #126: [Q] License of the repository

vicocz, 2024-11-01:
Hi,
what is the license of this repository please? I was not able to find any info if there is a default one due to the fact the repository is public.

https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository

imurvai, 2024-11-02:
Hi, to be honest I don't have any idea :) I just created it as public. Do you know where to check it?

vicocz, 2024-11-02:
I guess I can google only as well: https://opensource.guide/legal/#which-open-source-license-is-appropriate-for-my-project
I saw that e.g. pybricks has [MIT license](https://github.com/pybricks/pybricks-api/blob/master/LICENSE) as well.