imurvai/brickcontroller2 #135: Circuit Cubes with Android

s-hell, 2024-12-27:
We are trying to connect circuit cubes under Android.  They are getting connected and we see the three controls. 
But there is no reaction to connected motors when using the controls.

We tried different Versions. The current version 3.4 has the same problem. Any idea how to debug?

vicocz, 2024-12-31:
@s-hell Could you share the screenshot of Device test page to see versions of hardware please?
I can check mine in the menawhile once it get charged.

vicocz, 2025-01-01:
Mine is FW: SDK 47, HW: Tenka001 Cubes1.
Running app 3.4, running on Android 14.
Everything seems to be running fine.

vicocz, 2025-01-12:
Just possible explanation as recently, I've fixed (or at this time it's #138, the second commit) issue on iOS when any output command sent to CircuiCubes was not applied but the same app worked well on Android and Windows. Obvisously iOS was sensitive to the fact it was send using write with response but the characteristcs states it supports simple write only.

Simply said, once the app with this fix is release, please try again and let me know.

v5k248, 2025-06-23:
Just a quick note about this. I have 3 Circuit Cubes with different firmware versions. One was working fine on Android 10 with BrickController2 3.4. The other two behaved as described in this issue. I just upgraded to the [2025.3 release](https://github.com/vicocz/brickcontroller2/releases/tag/2025.03) from vicocz's branch and the ones that had a problem now work fine. 

@vicocz Thanks a lot for keeping the app updated in the branch! After installing your release I noticed a bunch of cool updates, looking forward to checking them out!