imurvai/brickcontroller2 #140: Mould King H4.0 support

streetjumper16, 2025-02-04:
Hello, is it possible to integrate the MK H4.0? the MK4.0 works, unfortunately not the MK H4.0 module.

vicocz, 2025-02-09:
Hi, AFAIK there is no support for any MouldKing in this app.

There is another fork of this app which might support some MKs, have a look there: https://github.com/J0EK3R/brickcontroller2

J0EK3R, 2026-01-31:
> Bonjour a tous je trouve pas comment faire pour l'application brickcontroler pour mould king MO-0019 merci d'avance a celui qui veut bien m'aider bonne journée à vous

It's the MK6.0 in the MO-0019.
It's supported by vicocz fork:
https://github.com/vicocz/brickcontroller2

Here is aclink to the releases:
https://github.com/vicocz/brickcontroller2/releases

jeremy230186, 2026-01-31:
> Open the releases page:
> https://github.com/vicocz/brickcontroller2/releases
>
> Choose one - latest release is 2025.05 (the newer once are pre-releases)
> Open Assets and download the file with ending ".apk" (for android).
>
> And install that on your Android device.
>
> Here is the direct link:
> https://github.com/vicocz/brickcontroller2/releases/download/2025.05/BrickController2_2025.05.apk

Je vous remercie énormément sa a fonctionné merci mille fois bonne soirée à vous

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>