imurvai/brickcontroller2 #141: Support for controller when app is running in background

LarsDD0815, 2025-02-12:
**Problem**
I have to keep my phone unlocked when using the game controller to control my model.

**Motivation**
I'd like to let my kids play using the controller/app but this is practically impossible, since my kids then would need to keep the phone unlocked (in their pocket) all the time.

**Request/Idea**
It would be great, if the app would continue to operate while the phone is locked. Thus, I could initially connect controller & lego-hub but then put the phone in the pockets, while playing using the controller.

**Plattforms**
Android & iOS - preferred ;)

ladi-j, 2025-02-19:
I believe this is a hard limitation (at least on iOS) due to the gamepad controller not working on the locked screen.

LarsDD0815, 2025-02-20:
OK. I also did some research and experiments which lead me to the same conclusion.

Thanks anyway :)

closed 2025-02-20 by LarsDD0815