imurvai/brickcontroller2 #144: Occasional 180 degree flip of center-position (BuWizz 3 + Lego PU center steered)

LarsDD0815, 2025-02-20:
I use a BuWizz 3.0 Pro with a Lego PU L-Motor as a steering-servo.
Sometimes - I could not identify a pattern yet - the servo goes wild and turns to a 180 degree angle - opposite of the previous center position. It then stays there, as if it was its new center-position. I can continue to control the servo from this new center position.

Anyone else had this issue?

I looked into the code and did not find anything wrong. The only idea I see is, that somehow the device reconnects, thus, sets some of its initial values wrong?

Currently the app is not usable for me, since the bug constantly breaks my steering-gears :(

Appreciate ideas/help!

Lars

vicocz, 2025-03-03:
Is it Android app please? if YES, what version of Android do you have please?

LarsDD0815, 2025-03-03:
Hey,
i do not know, if that happens on Android. I Used the 3.1 and the 3.4-release version and deployed them to my iPhone.

**Remark**
Version 3.4 did not work on iOS anymore. The BuWizz 3 did connect but motors/servos did not respond to any input from the app. I had to change class "BluetoothLEDevice":
`_peripheral.WriteValue(nativeData, nativeCharacteristic, CBCharacteristicWriteType.WithoutResponse);`
to
`_peripheral.WriteValue(nativeData, nativeCharacteristic, CBCharacteristicWriteType.WithResponse);`
I'm completely new to MAUI and BLE-connections so I did not investigate, why this change "worked" - probably you have an idea. 

**Wild guess regarding cause for my problem**

I tested the BuWizz with the original app and used the same tuning parameters (Kp, Ki, D_LP, ...). 
There is one big noticeable difference: the BuWizz app has a noticeable latency between action on the controller and reaction of the BuWizz brick. BUT: I could not reproduces the 180-degree flip with the original app. On the other hand: Brick Controller is super responsive. Just perfect. 
Maybe the problem is related to a higher signal-density, i.e. Brick Controller sends its signals with a higher "refresh rate" than the native app does?! Could that have side-effects in the BrickController-internal Firmware that lead to my problem? Wild guess...

vicocz, 2025-03-05:
Hi,
AFAIK BuWizz3 does not work on iOS due to the fact that commands are written without response, but the GATT characteristics states it requires write with response and based on my testing, iOS is sensisitive to that. Basically exactly the root cause you have applied.
I've applied fix in the fork I maintain for myself: https://github.com/vicocz/brickcontroller2/pull/88/files

LarsDD0815, 2025-03-12:
Found and fixed the issue. See: https://github.com/vicocz/brickcontroller2/pull/110